ouster-lidar/ouster-sdk #233: Error if no metadata file provided in replay mode

Chambana, 2021-03-19:
The docs indicate that if you don't provide a metadata json file for replay mode, the ros node will default to 1024x10.  However, I actually get an error that the node can't find the metadata file. 
[image]

[image]

kairenw, 2021-04-20:
Hi @Chambana, 

Unfortunately that bit of the README is outdated. We will be updating it shortly. You should always provide a metadata file when replaying data. If you don't have one, you can obtain one by following the instructions [here](https://github.com/ouster-lidar/ouster_example/tree/master#recording-data) for recording data and copying and saving the associated metadata file.

kairenw, 2021-04-27:
Hi @Chambana,

We've updated the README to reflect the necessity of a metadata file when replaying data. Thank you for poniting this out. Please let us know if you have difficulty obtaining a metadata file. I'm closing this ticket for now since it is resolved. 

Thanks!

closed 2021-04-27 by kairenw