ouster-lidar/ouster-sdk #234: Range map in ROS image node

theNded, 2021-03-25:
The point's property is defined as https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/include/ouster_ros/point.h#L25, 
with `range` as the 7-th property. However, in function `scan_to_cloud`, this field is filled with `REFLECTIVITY`:
https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/ros.cpp#L86

Could this be a bug, or is there any other specific reason for this implementation?
Thanks!

MaxandreOgeret, 2021-04-13:
Definitely a bug @dgarbuzov

kairenw, 2021-04-27:
Thanks for catching this, @theNded! We've fixed this in the [latest master](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/ros.cpp#L86), so I'm closing this issue now.

closed 2021-04-27 by kairenw