ouster-lidar/ouster-sdk #235: black strip on Ouster lidar

ge25nab, 2021-04-01:
[image]
[image]

I'm facing the black strips problem on ouster lidar, as shown in the first figure.  But it is ought to be the form in the second figure, where no black strips inside.
If someone faced similar problems or have some ideas about how to solve it, please leave messages here.

rcabg, 2021-04-15:
hey @ge25nab , did you compile using RELEASE mode?

catkin_make -DCMAKE_BUILD_TYPE=Release

kairenw, 2021-08-16:
Hi @ge25nab, if this is still an issue, I'd encourage you to [contact Ouster support ](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86). Closing this issues now.

closed 2021-08-16 by kairenw