ouster-lidar/ouster-sdk #237: Rviz not showing anything & Receiving empty topics despite successful connection

dlee640, 2021-04-12:
Hello.

My system is: Ubuntu 18.04
My lidar is: OS0-32 with newest firmware: 2.0

I was able to successfully visualize the sensor stream using sample client visualizer. Both the image & lidar point clouds were nicely displayed.

Yet, when I try to connect to sensor using `roslaunch ouster_ros ouster.launch sensor_hostname:=os-992106000190.local udp_dest:=192.254.192.171 lidar_mode:=1024x20 viz:=true`, I get following results:

```
dlee640@dlee640-T580:~/ouster_example/build/ouster_viz$ roslaunch ouster_ros ouster.launch sensor_hostname:=os-992106000190.local udp_dest:=192.254.192.171 lidar_mode:=1024x20 viz:=true
... logging to /home/dlee640/.ros/log/fda50dc8-9bc6-11eb-aae3-b46bfc2dbb2b/roslaunch-dlee640-T580-13222.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dlee640-T580:43407/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 1024x20
 * /os_node/lidar_port: 0
 * /os_node/metadata: 
 * /os_node/replay: False
 * /os_node/sensor_hostname: os-992106000190.l...
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 192.254.192.171
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
    viz_node (ouster_ros/viz_node)

auto-starting new master
process[master]: started with pid [13320]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fda50dc8-9bc6-11eb-aae3-b46bfc2dbb2b
process[rosout-1]: started with pid [13331]
started core service [/rosout]
process[os_node-2]: started with pid [13337]
process[os_cloud_node-3]: started with pid [13339]
process[viz_node-4]: started with pid [13344]
[ INFO] [1618256453.272928622]: Client version: 0.2.0-dev0+feb4f19-release
[ INFO] [1618256453.273559648]: Connecting to os-992106000190.local; sending data to 192.254.192.171
[ INFO] [1618256453.273586251]: Waiting for sensor to initialize ...
process[relay_lidar-5]: started with pid [13346]
process[relay_imu-6]: started with pid [13356]
[ INFO] [1618256462.587816260]: Sensor initialized successfully
[ INFO] [1618256462.588745577]: Wrote metadata to $ROS_HOME/os-992106000190.local.json
[ INFO] [1618256462.589817073]: Using lidar_mode: 1024x20
[ INFO] [1618256462.589893523]: OS-0-32-U1 sn: 992106000190 firmware rev: v2.0.0
```

**Important thing to note is that sensor is connected successfully ( I can hear the sensor running as I roslaunch this command) and I see no particular error messages. Yet, I see nothing on my rViz window.

Also, when I type rosnode list & rostopic list, I get following. As you can see, nodes & topics are identified, but no messages are actually coming thru, which explains why rViz was empty. Anyone had this issue? I have no problem with sample visualizer, so I think this is ROS issue, not the connectivity issue. Maybe firmware 2.0 is not compatible?? I am lost!** :

`dlee640@dlee640-T580:~$ rostopic list
/os1_node/imu_packets
/os1_node/lidar_packets
/os_cloud_node/imu
/os_cloud_node/points
/os_node/imu_packets
/os_node/lidar_packets
/rosout
/rosout_agg
/tf_static
dlee640@dlee640-T580:~$ rosnode list
/os_cloud_node
/os_node
/relay_imu
/relay_lidar
/rosout
/viz_node
dlee640@dlee640-T580:~$ rostopic hz os_node/imu_packets
subscribed to [/os_node/imu_packets]
no new messages
no new messages
no new messages
^Cno new messages
`

kwong0, 2021-04-16:
Hi @dlee640,

The `ouster_ros` code should be fw 2.0 compatible, so it is likely a networking issue of some sort. If you got the sample client visualizer working, I would double check the `udp_dest` you used.

To debug locally, you can try confirming that the packets are arriving to the appropriate IP and port. 

To find the appropriate ports, connect your sensor, try to run the visualizer, and then shut it down. Then:
1. nc to the sensor: `nc os-992106000190.local 7501`
2. `get_config_param active` once you're in
3. Check the result fo the configured ports `udp_port_lidar` and `udp_port_imu`. They should be random.

To check packets, you can use Wireshark or the following:
`nc -ul <PORT> | od -b | head`.

If you're still having issues after this, let us know, and we can put you in touch with our FAEs. 

Oh - and the visualizer launched from `ouster.launch` is actually not RViz, by the way! It launches a ros node which uses the same visualization library as our sample client visualizer. If you want to use RViz, there is a .rviz inside `ouster_ros` as well. (However it is unlikely to solve your current bug since it seems that your networking is not configured properly).

dlee640, 2021-04-20:
Thank you for your response @kwong0 but these things did not work. Actually, the ouster.launch was successful in running the visualization, as well as I get response from /os_cloud_node.

In short, only 1/3 of the nodes seem to be working correctly, and I am still not certain on why so. I have modified my lidar port number to 7502 and imu port number to 7503 as modified on lot of y'alls launch file.

To sum up, out of the following topics, 
```
slambox@slambox-NUC10i7FNH:~/ouster_ws$ rostopic list
/os1_node/imu_packets
/os1_node/lidar_packets
/os_cloud_node/imu
/os_cloud_node/points
/os_node/imu_packets
/os_node/lidar_packets
```

I ONLY get the response & data from:
```
/os_cloud_node/imu
/os_cloud_node/points
```
Not published response( `slambox@slambox-NUC10i7FNH:~/ouster_ws$ rostopic hz os1_node/lidar_packets WARNING: topic [/os1_node/lidar_packets] does not appear to be published yet` ) from:
```
/os1_node/imu_packets
/os1_node/lidar_packets

```

Message built error ( `slambox@slambox-NUC10i7FNH:~/ouster_ws$ rostopic hz os_node/lidar_packets ERROR: Cannot load message class for [ouster_ros/PacketMsg]. Are your messages built? `)from:
```
/os_node/imu_packets
/os_node/lidar_packets
```

kairenw, 2021-04-20:
Hi @dlee640,

Sorry for the misunderstanding. I think I understand now -- your launch file works to launch the sample visualizer, but when you run rviz nothing appears. 

Have you built the ROS code? It's actually a separate [section ](https://github.com/ouster-lidar/ouster_example#example-ros-code)of the README. That should fix the errors about whether your  messages are built.

After you fix that, you should receive messages on the `os_node` and `os_cloud_node` topics, but you still should not be receive anything on the `os1_node` topics since those are just relayed topics, in case you are playing from a bag with recorded data from firmware 1.13

kairenw, 2021-06-11:
Closing due to lack of response. Feel free to re-open if this continues being an issue.

closed 2021-06-15 by kairenw

j2856256, 2022-06-15:
我目前有一樣的問題，想請問解決了嗎？

j2856256, 2022-06-15:
我現在有同樣的問題，想問一下解決了嗎？

kairenw, 2022-06-15:
Hi @j2856256,

This issue is currently closed. Could you open a new ticket and describe your problem in detail? Thanks!