ouster-lidar/ouster-sdk #240: Failed to initialize sensor at : os1-99XXXXXXXXXX.local

bigbigpark, 2021-05-04:
Hi guys! I have OS1 lidar. I got some troubles :(

`ping -4 -c4 os1-99XXXXXXXXXX.local`  => success  // os1 host name
`ping -4 -c4 192.168.0.111`  => success  // My PC ip

I tested ping for both sensor and PC.
but when I run
`roslaunch ouster_ros ouster.launch os1_hostname:=192.168.0.111`

I got trouble like that images.

Anyone who knows the solution?

[image]

bigbigpark, 2021-05-04:
I solved this problem by updating firmware verison.

kairenw, 2021-05-14:
Glad it was solved!

RFRIEDM-Trimble, 2021-06-03:
I'm seeing this on v2.0 FW.

dgarbuzov, 2021-06-03:
Are you able to ping the sensor and/or load the web UI? Is your sensor in STANDBY mode?

RFRIEDM-Trimble, 2021-06-03:
Ping: Yes
Avahi-resolve with hostname: Yes
Web UI: Yes
STANDBY: No. It works fine with the address, just not the hostname. I hear it spinning now 
Netcat: Yes

dgarbuzov, 2021-06-03:
> It works fine with the address, just not the hostname
just to clarify: running the launch file using the IP works, it's just using the hostname that's failing? And you're able to ping the sensor using the same hostname.

Are you connecting through a router, running a dhcp server locally, or using mdns?

RFRIEDM-Trimble, 2021-06-03:
I ran into this issue in the ROS2 repo. I don't use ROS1, but to verify, checked the client libraries here with `ouster_client_example`. In the ROS2 param file, I can use a direct address, but not a hostname.

Same behavior shown below here, showing it's not an issue with ROS2.

```
/ouster_client_example 192.168.90.2 7600
Ouster client example 0.2.0-dev0
Connecting to "192.168.90.2"... ok
Gathering metadata...
  Firmware version:  v2.0.0
  Serial number:    XXXX
  Product line:      OS-0-128
  Scan dimensions:   512 x 128
  

/ouster_client_example os-XXX.local 7600
Ouster client example 0.2.0-dev0
Connecting to "os-xxx.local"... getaddrinfo: Name or service not known
Failed to connect

  ```
  
  I have static IPv4 configured on both the sensor and my machine, directly connected. For IPv6, I can use link local addressing with the scope ID to connect. I know link local addressing on IPv6 can cause issues if some library truncates the scope ID out of the domain name resolution, such as currently seen with [avahi](https://github.com/lathiat/avahi/issues/110). The call to getaddrinfo seems to be the problem.  
  
$  avahi-resolve -n os-xxx.local
os-xxx.local	fe80::be0f:a7ff:fe00:2861

dgarbuzov, 2021-06-03:
If you're using avahi and libnss-mdns, I think the defaults in `/etc/nsswitch.conf` on Ubuntu use `mdns4_minimal` which won't resolve ipv6 addresses.

RFRIEDM-Trimble, 2021-06-03:
I am running this inside a ROS2 Foxy docker container. Avahi is not installed inside the supplied docker container. Looking at `/etc/nsswitch.conf`, I see this:
```
# cat /etc/nsswitch.conf 
# /etc/nsswitch.conf
#
# Example configuration of GNU Name Service Switch functionality.
# If you have the `glibc-doc-reference' and `info' packages installed, try:
# `info libc "Name Service Switch"' for information about this file.

passwd:         files
group:          files
shadow:         files
gshadow:        files

hosts:          files dns
networks:       files

protocols:      db files
services:       db files
ethers:         db files
rpc:            db files

netgroup:       nis
```

Adding `mdns4_minimal` has no changes. 

Also tried this: `hosts:          files mdns4_minimal [NOTFOUND=return] dns mdns4`

From the [documentation](http://0pointer.de/lennart/projects/nss-mdns/), there doesn't seem to be a need to restart a service or something after this file is changed, but anything I throw in the file has no visible change on behavior. 

 Seems like there's some issue with mdns inside the container that needs attention that may be causing this.

RFRIEDM-Trimble, 2021-06-03:
I just built `ouster_client_example` outside of the ROS2 Foxy docker on my 18.04 host and it works fine with hostname resolution. 
My host has the following config with nsswitch:
`hosts:          files mdns_minimal [NOTFOUND=return] dns myhostname` and can resolve os-###.local.

dgarbuzov, 2021-06-03:
Oh, yeah docker throws in a whole extra layer of fun. Agreed re: mdns + ipv6 link-local addresses being easiest, and that's actually what I use day-to-day. My setup:

- host system running systemd-resolved, with systemd managing /etc/resolv.conf as described [here](https://wiki.archlinux.org/title/Systemd-resolved)
- mdns enabled on the interface connected to the sensor via `resolvectl`
- docker containers run with `--network=host`, so they can reach the stub listener on the host system

If you're not using systemd-resolved on the host system and have to use avahi, you might have some luck with bind-mounting the avahi daemon socket and configuring libnss-mdns inside the container. IIRC I gave up on that approach since it requires some configuration per-container, which gets old pretty quick.

closed 2021-06-03 by dgarbuzov