ouster-lidar/ouster-sdk #241: how do i get the cameraInfo for the images in the img_node.cpp?

Sagexs, 2021-05-06:
hi,

I would like to use the ambient_image for an algorithm that also needs the camera_info. Can someone tell me how to calculate that for this sensor?  

I now the real position of each pixel and could use an optimization algorithem to comput the projection and camera matrix, but i think there is a way thats far easyer, but can't see it right now.

please help me out a little bit.

Greadings

peci1, 2022-07-01:
Unless your sensor is stationary, treating the sensor as a camera is not a good way. If the sensor moves during the 100 ms that one scan takes, the projection center is moving, thus invalidating all the assumptions made when using projective camera model.

Samahu, 2022-08-11:
Not sure if I understood your question or what you mean by camera_info in this context. Do you mean the sensor extensics?

closed 2022-08-11 by Samahu