ouster-lidar/ouster-sdk #242: json/json.h No such file or directory

mswish, 2021-05-06:
I am on a fresh install of Ubuntu. After following the directions for building ROS code I receive this error:
```In file included from /home/swisher/code/myworkspace/src/ouster_example/ouster_ros/src/os_node.cpp:20:
/home/swisher/code/myworkspace/src/ouster_example/ouster_ros/../ouster_client/include/ouster/types.h:8:10: fatal error: json/json.h: No such file or directory
    8 | #include <json/json.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
```
```
In file included from /home/swisher/code/myworkspace/src/ouster_example/ouster_ros/../ouster_client/include/ouster/client.h:12,
                 from /home/swisher/code/myworkspace/src/ouster_example/ouster_ros/src/img_node.cpp:25:
/home/swisher/code/myworkspace/src/ouster_example/ouster_ros/../ouster_client/include/ouster/types.h:8:10: fatal error: json/json.h: No such file or directory
    8 | #include <json/json.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
```
The command `sudo apt install libjsoncpp-dev` finishes without problem. 
OS: Ubuntu 20.04.2 LTS
Shell: Bash
Libjsoncpp-dev version: 1.7.4-3.1ubuntu2
ROS Version: noetic

NovoG93, 2021-05-12:
Same problem here, also on a fresh install of 20.04 with ROS noetic.

matlabbe, 2021-05-13:
Remove this line: https://github.com/ouster-lidar/ouster_example/blob/e1176f427f68eb0807bb15ccabe34aea47a1c5d3/ouster_client/include/ouster/types.h#L8

kairenw, 2021-05-15:
Thanks for catching this! Should be addressed on master now.

closed 2021-05-15 by kairenw