ouster-lidar/ouster-sdk #247: Point Definition in Ouster ROS Example

wangziqi000, 2021-05-19:
Hi,

I noticed in the ROS Example, the os_cloud_node processes LiDAR packets into LiDAR scans, and then convert to them into a point cloud. The points in the PointCloud2D is defined as follows:

struct EIGEN_ALIGN16 Point {
    PCL_ADD_POINT4D;
    float intensity;
    uint32_t t;
    uint16_t reflectivity;
    uint8_t ring;
    uint16_t ambient;
    uint32_t range;
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
};

**I was wondering about the physically meaning of those fields.**  

The closest information I found is in the Software User Manual, wherein the packet definition part four concepts are introduced: Range(Distance), Signal Photons(Intensity), Reflectivity, and NIR Photons (Ambient Light Intensity). 

However, I am confused by the aforementioned structure design:
(1) PCL_ADD_POINT4D should include x,y,z plus intensity, but why is there a separate intensity field defined?
(2) How should I interpret these fields: t, ring, and range?

Any help is greatly appreciated!

Sincerely

dgarbuzov, 2021-05-19:
`t` is the the timestamp of the point, in nanoseconds *relative to the timestamp of the point cloud message*
`ring` is the index of the lidar channel that took the measurement, with 0 referring to the top-most (highest altitude angle) beam
`range` is the measured range, in mm, from which the cartesian coordinates were computed

Re: POINT4D -- the 4th field is padding, not intensity:
https://pcl.readthedocs.io/projects/tutorials/en/master/adding_custom_ptype.html#what-pointt-types-are-available-in-pcl

Hope that helps!

wangziqi000, 2021-05-21:
Thank you! that is super helpful!

closed 2021-05-26 by kairenw