ouster-lidar/ouster-sdk #249: option for 16-bit depth image output

Chambana, 2021-05-26:
The Ouster ROS depth image is scaled to only 8-bit depth values (mono8), but I believe the sensor hardware is capable of providing 16-bit depth.  Unfortunately, my application requires 16-bit depth data similar to what's provided by a realsense camera.  Can you consider this as a feature request (outputting a 16-bit depth image in ROS) or is there a technical limitation here?

Chambana, 2021-05-27:
The sensor range value is actually 20bits of resolution packaged in a 32-bit type

[image]

dgarbuzov, 2021-05-28:
Yep, no reason not to output 16-bit images. Think the current implementation was just a quick demo for output in rviz, I think. Can you talk more about how you intend to use the ROS image topics? To fit range into 16 bits we usually round to 4mm and throw out returns beyond ~260m. Would that work for you?

Chambana, 2021-05-28:
Awesome.  Sure, the use case for the depth images is to feed a planar region detection algorithm -- so we're looking for the highest resolution possible for very close-range obstacles (a few meters away).  

For our use case, imagine the OS0-128 positioned at human head height and down-angled at a pile of random cinder blocks at your feet.  We need the depth map resolution to precisely capture the ranges of those cinder blocks/ground plane/etc, so that we can calculate a proper foothold.  For long range (you mention 260m :) ), we don't really care about depth resolution at that range.  

I think your proposed solution would be a great improvement for us.  Is there a implementation that would preserve the 3mm of resolution vice rounding to 4mm (and still fit within 16-bits)?

dgarbuzov, 2021-05-28:
Sure -- if you just mask out values over 2^16 you can keep 65 m of range without rounding. Not sure we'll go as far as making this configurable, but it should be a matter of commenting out one line.

Chambana, 2021-05-28:
That would be great.  Thanks for knocking this out -- our team is definitely looking forward to this update!

I'll be checking this Issue for any updates on implementation (or timeline).

dgarbuzov, 2021-06-08:
should be addressed by #259

closed 2021-06-08 by dgarbuzov