ouster-lidar/ouster-sdk #250: Is there a way to if a lidar beam has no return?

versatran01, 2021-05-27:
For example, in an open space where everything is very far away, what would the range value be in the packet?

versatran01, 2021-05-27:
Related ROS REP
https://www.ros.org/reps/rep-0117.html

dgarbuzov, 2021-05-28:
Currently Ouster sensors output (and the sample code here passes through) a zero range when there's no return.

Thanks for linking the REP, hadn't seen that before. Do you have an example of a lidar or rgbd camera driver that follows this spec for PointCloud2 topics? It's not currently possible to distinguish the "too close" vs "out of range" cases so it's not obvious how we'd implement this, but maybe there's an opportunity to do something more standard.

versatran01, 2021-05-28:
Thanks for the reply. So for gen 2 sensor, 0 means "out of range", anything >0 should be treated as a hit, is this correct?

i-sheremet, 2021-06-17:
> It's not currently possible to distinguish the "too close" vs "out of range" cases so it's not obvious how we'd implement this, but maybe there's an opportunity to do something more standard.

@dgarbuzov Do you know when it's planned to make it possible to distinguish between those two?

mortenfyhn, 2021-06-18:
Regarding "too close" vs. "out of range": The OS1 datasheet mentions

> 0 m - 0.3 m blockage detection flag to indicate object within 0.3 m (v2.0 beta feature)

Doesn't that mean that it should be able to detect "too close" as something different from "out of range"?

versatran01, 2021-06-18:
Where is this flag?

i-sheremet, 2021-06-25:
Does the new Firmware 2.1.1 solves this issue?

dgarbuzov, 2021-07-01:
Sorry for the late response:

> Just to confirm, 0 means "out of range", anything >0 should be treated as a hit, is this correct?

@versatran01: 0 means there is no valid range reading (the way conversion from raw range readings to coordinates in the lidar frame is defined, this would actually be *inside* the sensor). That could either be out of range, within the minimum range, or just a missed detection.

@i-sheremet I can't comment on feature timelines, but I can confirm that the general laser link above isn't accurate, but for further questions about min/max range detection, please contact Ouster support.

closed 2021-07-01 by dgarbuzov

versatran01, 2021-07-01:
May I ask again where is the block detection flag?

mortenfyhn, 2021-07-02:
It would be really nice to have this flag made available in the point clouds created by this ROS driver.

peci1, 2021-09-10:
> Benjamin Freret commented:
> Hello Martin,
> Thanks for reaching us. This flag is not available, even though it is said in the documentation that it is.
> Apologies for this inconvenience,
> Benjamin, Customer Success Engineer

shubham-shahh, 2022-05-09:
> > Benjamin Freret commented:
> > Hello Martin,
> > Thanks for reaching us. This flag is not available, even though it is said in the documentation that it is.
> > Apologies for this inconvenience,
> > Benjamin, Customer Success Engineer

is this flag available now?