ouster-lidar/ouster-sdk #251: Working with multicast configured units?

canvasus, 2021-05-27:
Is there a way to connect to multicast-configured OSx units with the Python SDK?

dgarbuzov, 2021-05-28:
Setting "udp_dest" to a multicast address with `client.set_config` and listening for packets seems to work for me. Are you running into any particular issues or looking for more documentation?

canvasus, 2021-06-03:
Ok thanks. On our system I could not initialize the clients. I will try this again and come back with more details.

For clarity: did you test on Windows or Linux, and do you think there could be different behaviour?

dgarbuzov, 2021-06-03:
I tested on linux. I'd be surprised if multicast UDP behaved differently on Windows, but I could be wrong.

Re failure to initialize: that happens when the client is unable to read metadata from the sensor over TCP, so it might be unrelated to the udp_dest setting.

closed 2021-07-01 by dgarbuzov

dgarbuzov, 2021-08-23:
I think I was confusing multicast with broadcast addresses. Feel free to reopen