ouster-lidar/ouster-sdk #256: Cartographer ROS with Ouster Lidar: Queue waiting for data: (0, points2)

asable67, 2021-06-04:
Hello,

I am attempting to implement Cartographer with a RealSense camera and an Ouster OS0-64 sensor for a 3D mapping application, but I am struggling with the launch and configuration files. I appear to be encountering an issue with the lidar points somewhere along the way. Below is the output in the terminal after running the launch file <code>rs_and_os_driver.launch</code>:

<pre><font color="#4E9A06"><b>spike@slam</b></font>:<font color="#3465A4"><b>/media/spike/Cartographer/OusterandROSStuff</b></font>$ ./launchros_sccartographer.sh 
... logging to /home/spike/.ros/log/ceda512e-c3b2-11eb-ad83-0242a1c82dcc/roslaunch-slam-7041.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

<b>started roslaunch server http://slam:41927/</b>

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 62
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: t265
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 800
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 848
 * /camera/realsense2_camera/gyro_fps: 200
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 278 lines cut ...]
 * topic: /rosout
    * to: /rosout
    * direction: outbound (44877 - 127.0.0.1:40548) [10]
    * transport: TCPROS
 * topic: /os_cloud_node/points
    * to: /cartographer_node
    * direction: outbound (44877 - 127.0.0.1:41226) [16]
    * transport: TCPROS
 * topic: /tf_static
    * to: /cartographer_node
    * direction: outbound (44877 - 127.0.0.1:41232) [11]
    * transport: TCPROS
 * topic: /os_node/lidar_packets
    * to: /os_node (http://slam:43589/)
    * direction: inbound (38028 - slam:44097) [15]
    * transport: TCPROS
 * topic: /os_node/imu_packets
    * to: /os_node (http://slam:43589/)
    * direction: inbound (38032 - slam:44097) [17]
    * transport: TCPROS
</pre>

The relevant launch files, configuration file options, and tf tree are [posted here](https://github.com/asable67/Ouster-Lidar-Cartographer-ROS.git). If there is any additional information or files I can provide, please don't hesitate to ask, as I am new to ROS and Cartographer and do not know what information is most pertinent.

Thanks!

asable67, 2021-06-24:
After trying many things, I finally solved this by changing the ouster sensor's time synchronization mode from the default internal oscillator to time from ptp1588, as described in the Ouster Software user manual... Not sure why this was the solution, but Cartographer now works and does real-time and offline SLAM.

closed 2021-06-24 by asable67

juliangaal, 2022-04-08:
> After trying many things, I finally solved this by changing the ouster sensor's time synchronization mode from the default internal oscillator to time from ptp1588, as described in the Ouster [...]

ROS cannot handle the sensor beginning at timestamp 0, while all other sensors and sotfware run with ROS::Time