ouster-lidar/ouster-sdk #257: TCP configuration can hang with no warning

RFRIEDM-Trimble, 2021-06-08:
As found in https://github.com/ros-drivers/ros2_ouster_drivers/issues/72, during the TCP configuration phase, the calls to `recv` inside of `do_tcp_cmd(..)`  operate on a blocking socket. 
The current behavior of my OS0-128, running V2.0.0 firmware, is to accept TCP connections from multiple clients but only serve the oldest client data until that client disconnects. Thus, if you somehow run multiple drivers, the ouster is unresponsive to TCP commands. 
The existing driver blocks on TCP commands waiting for a response, and will sit there endlessly until the ouster responds. This is not ideal, as found in ROS2, it prevents data being published, but the node appears to be running.

I propose adding an optional timeout to prevent blocking indefinitely. This could be done to preserve existing blocking behavior.

dgarbuzov, 2021-06-30:
The sensor interface behavior is intentional -- I'll see if we can make sure it's documented in the next update. I think it was expected that well-behaved clients would close the TCP socket after sending a series of commands, but I could see how someone might want to leave it open to poll for alerts, for example.

Yes, we need a timeout on the `recv()` to deal with this a bit more gracefully. I'm not sure whether it needs to be configurable. Hypothetically, someone could be relying on the indefinite blocking by keeping a socket open and queuing up commands from other threads, but that seems pretty unlikely, no?

We can discuss further on #258

RFRIEDM-Trimble, 2021-06-30:
The behavior of the ouster client code keeps the TCP connection open for the duration the executable is running.

See my PR for my implementation of a fix for this.

dgarbuzov, 2021-06-30:
> The behavior of the ouster client code keeps the TCP connection open for the duration the executable is running.

Do you mean the ROS2 driver? Or is there another bug fixed I didn't see in the diff. I believe the client library closes the socket reliably after sending any commands over TCP. E.g. after the client is initialized here:

https://github.com/ouster-lidar/ouster_example/blob/b8b23c35d7b719d69341a438d386a801688aa6a4/ouster_client/src/client.cpp#L572

kairenw, 2021-08-16:
Should be addressed by PR #275. Closing issue.

closed 2021-08-16 by kairenw

RFRIEDM-Trimble, 2021-08-16:
LGTM