ouster-lidar/ouster-sdk #260: picking up specified frame from pcap file

kazu0622, 2021-06-10:
I'm trying to design python app to retrieve frame data from pcap file.
The frames I got seem to be skipy despite I gave consecutive number for scans.

I made a copy  your sample code and modified like this:
metadata = client.SensorInfo(metadata_path)
source = pcap.Pcap(pcap_path, metadata)
scans = client.Scans(source)
# iterate `scans` and get the 84th LidarScan
for i in range(0,100)
    scan = nth(scans, i)
    ranges = scan.field(client.ChanField.RANGE)

I'd like to know where the problem is.

I'll attach 2 image files I got.
[image]
[image]

Thanks.

kairenw, 2021-06-10:
Hi @kazu0622,

You're skipping the scans in between by using [`nth`](https://more-itertools.readthedocs.io/en/stable/api.html#more_itertools.nth). I would try replacing your loop with `for scan in scans` to iterate through your pcap.

kazu0622, 2021-06-11:
Hi @kairenw,
Thank you for your quick response and I appreciate your tips.
I was able to solve this issue.

Thanks.

closed 2021-06-11 by kazu0622