ouster-lidar/ouster-sdk #261: ROS KINETIC:     udp socket(): Address family not supported by protocol

AdriannaGmz, 2021-06-14:
Hi!, I'm trying to run the example on Ubuntu 16, ROS Kinetic but when I try roslaunch, it can not start:

`administrator@cpr-a200-0653:~/ws/t2_ws$ roslaunch ouster_ros ouster.launch sensor_hostname:=192.168.132.30 udp_dest:=192.168.132.60 metadata:=/home/administrator/ISA4/tmp/met.json lidar_mode:=512x10 viz:=false
... logging to /home/administrator/.ros/log/83a905c2-ccfb-11eb-8955-0016d33011c0/roslaunch-cpr-a200-0653-4354.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.132.60:36035/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 512x10
 * /os_node/lidar_port: 0
 * /os_node/metadata: /home/administrat...
 * /os_node/replay: False
 * /os_node/sensor_hostname: 192.168.132.30
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 192.168.132.60
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)

ROS_MASTER_URI=http://192.168.132.2:11311

process[os_node-1]: started with pid [4372]
process[os_cloud_node-2]: started with pid [4373]
process[relay_lidar-3]: started with pid [4377]
process[relay_imu-4]: started with pid [4378]
[ INFO] [1623667460.312223866]: Client version: 0.2.0+b8b23c3-release
[ INFO] [1623667460.312296449]: Connecting to 192.168.132.30; sending data to 192.168.132.60
[ INFO] [1623667460.312320050]: Waiting for sensor to initialize ...
udp socket(): Address family not supported by protocol
udp socket(): Address family not supported by protocol
[ERROR] [1623667460.314070862]: Failed to initialize sensor at: 192.168.132.30
[ERROR] [1623667460.316414223]: Calling config service failed
================================================================================REQUIRED process [os_node-1] has died!
process has died [pid 4372, exit code 1, cmd /home/administrator/ws/t2_ws/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/administrator/.ros/log/83a905c2-ccfb-11eb-8955-0016d33011c0/os_node-1.log].
log file: /home/administrator/.ros/log/83a905c2-ccfb-11eb-8955-0016d33011c0/os_node-1*.log
Initiating shutdown!
================================================================================
[relay_imu-4] killing on exit
[relay_lidar-3] killing on exit
[os_cloud_node-2] killing on exit
[os_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete`

I have tried the same pkg in another computer with ROS Noetic and it works fine. For this setup, I need the machine with ROS Kinetic to get the lidar launch working. Would you have an idea on how to solve this? thanks in advanced.

dgarbuzov, 2021-06-30:
do you have ipv6 disabled via a kernel parameter? You can check by running `cat /proc/cmdline` and looking for `ipv6.disable=1`. To work around this, you can either re-enable ipv6 in the kernel (you can still disable it on the system using sysctl, or just on the particular network interface). Another option would be to turn off the dual-stack socket support here:

https://github.com/ouster-lidar/ouster_example/blob/b8b23c35d7b719d69341a438d386a801688aa6a4/ouster_client/src/client.cpp#L68

by replacing `AF_INET6` with `AF_INET`, though I haven't tried that out recently.

dgarbuzov, 2021-07-01:
After reviewing some older tickets, this is probably common enough that we should add support for linux systems with ipv6 disabled in the kernel.

dgarbuzov, 2021-08-03:
Should be fixed on master now

closed 2021-08-03 by dgarbuzov