ouster-lidar/ouster-sdk #262: Build error [-Werror=deprecated-copy]   on Ubuntu 18.04

alessandromorra, 2021-06-14:
When building on my machine I get the following error - any suggestions why?

```
alessandro@alessandro-MacBookPro:~/catkin_ws/src$ catkin build ouster_ros
---------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/alessandro/catkin_ws
---------------------------------------------------------------
Build Space:        [exists] /home/alessandro/catkin_ws/build
Devel Space:        [exists] /home/alessandro/catkin_ws/devel
Install Space:      [unused] /home/alessandro/catkin_ws/install
Log Space:          [exists] /home/alessandro/catkin_ws/logs
Source Space:       [exists] /home/alessandro/catkin_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------
[build] Found '3' packages in 0.0 seconds.                                                                                              
[build] Updating package table.                                                                                                         
Starting  >>> ouster_ros                                                                                                                
________________________________________________________________________________________________________________________________________
Warnings   << ouster_ros:cmake /home/alessandro/catkin_ws/logs/ouster_ros/build.cmake.000.log                                           
CMake Warning (dev) at /usr/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (eigen) does
  not match the name of the calling package (PCL).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:153 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:638 (find_eigen)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:850 (find_external_library)
  CMakeLists.txt:11 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PCL_COMMON)
  does not match the name of the calling package (PCL).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:911 (find_package_handle_standard_args)
  CMakeLists.txt:11 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:41 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
[... 56 lines cut ...]
[build]   Abandoned: None.                                                                                                              
[build]   Failed:    1 packages failed.                                                                                                 
[build] Runtime: 17.4 seconds total.                                                                                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them.
alessandro@alessandro-MacBookPro:~/catkin_ws/src$ 
```

System:

```
alessandro@alessandro-MacBookPro:~/catkin_ws/src$ g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

alessandro@alessandro-MacBookPro:~/catkin_ws/src$ gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

Many thanks for the support!

dgarbuzov, 2021-07-01:
That's strange -- I'm not getting these errors building in a fresh ros melodic container. Tried using `python-catkin-tools` like you did here as well.

Looks like the deprecations are coming from `ros-melodic-tf2-eigen`. If you could provide the output of `dpkg -l 'ros-melodic*'` maybe we can narrow down what the differences might be in our environments.

alessandromorra, 2021-07-06:
Dear @dgarbuzov 

Thanks for the support:

```
dpkg -l 'ros-melodic*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                           Version                      Architecture                 Description
+++-==============================================-============================-============================-==================================================================================================
ii  ros-melodic-actionlib                          1.12.1-1bionic.20210505.0259 amd64                        The actionlib stack provides a standardized interface for interfacing with preemptable tasks.
ii  ros-melodic-actionlib-msgs                     1.12.8-1bionic.20210505.0106 amd64                        actionlib_msgs defines the common messages to interact with an action server and an action client.
ii  ros-melodic-actionlib-tutorials                0.1.11-0bionic.20210505.0303 amd64                        The actionlib_tutorials package
ii  ros-melodic-angles                             1.9.12-1bionic.20201017.0325 amd64                        This package provides a set of simple math utilities to work with angles.
ii  ros-melodic-base-local-planner                 1.16.7-1bionic.20210505.0413 amd64                        This package provides implementations of the Trajectory Rollout and Dynamic Window approaches to l
ii  ros-melodic-bond                               1.8.5-1bionic.20210505.01084 amd64                        A bond allows two processes, A and B, to know when the other has terminated, either cleanly or by 
ii  ros-melodic-bond-core                          1.8.5-1bionic.20210505.02204 amd64                        A bond allows two processes, A and B, to know when the other has terminated, either cleanly or by 
ii  ros-melodic-bondcpp                            1.8.5-1bionic.20210505.02062 amd64                        C++ implementation of bond, a mechanism for checking when another process has terminated.
ii  ros-melodic-bondpy                             1.8.5-1bionic.20210505.02103 amd64                        Python implementation of bond, a mechanism for checking when another process has terminated.
ii  ros-melodic-camera-calibration                 1.15.0-1bionic.20210505.0408 amd64                        camera_calibration allows easy calibration of monocular or stereo cameras using a checkerboard cal
ii  ros-melodic-camera-calibration-parsers         1.11.13-0bionic.20210505.025 amd64                        camera_calibration_parsers contains routines for reading and writing camera calibration parameters
ii  ros-melodic-camera-info-manager                1.11.13-0bionic.20210505.030 amd64                        This package provides a C++ interface for camera calibration information.
ii  ros-melodic-catkin                             0.7.29-1bionic.20201014.1914 amd64                        Low-level build system macros and infrastructure for ROS.
ii  ros-melodic-class-loader                       0.4.1-0bionic.20201015.03300 amd64                        The class_loader package is a ROS-independent package for loading plugins during runtime and the f
ii  ros-melodic-clear-costmap-recovery             1.16.7-1bionic.20210505.0415 amd64                        This package provides a recovery behavior for the navigation stack that attempts to clear space by
ii  ros-melodic-cmake-modules                      0.4.2-0bionic.20201015.02171 amd64                        A common repository for CMake Modules which are not distributed with CMake but are commonly used b
ii  ros-melodic-common-msgs                        1.12.8-1bionic.20210505.0441 amd64                        common_msgs contains messages that are widely used by other ROS packages.
ii  ros-melodic-common-tutorials                   0.1.11-0bionic.20210505.0321 amd64                        Metapackage that contains common tutorials
ii  ros-melodic-compressed-depth-image-transport   1.9.5-0bionic.20210505.04390 amd64                        Compressed_depth_image_transport provides a plugin to image_transport for transparently sending de
ii  ros-melodic-compressed-image-transport         1.9.5-0bionic.20210505.04391 amd64                        Compressed_image_transport provides a plugin to image_transport for transparently sending images e
ii  ros-melodic-control-msgs                       1.5.1-1bionic.20210505.01261 amd64                        control_msgs contains base messages and actions useful for controlling robots.
ii  ros-melodic-control-toolbox                    1.18.2-1bionic.20210505.0309 amd64                        The control toolbox contains modules that are useful across all controllers.
ii  ros-melodic-controller-interface               0.18.3-1bionic.20210505.0210 amd64                        Interface base class for controllers.
ii  ros-melodic-controller-manager                 0.18.3-1bionic.20210505.0250 amd64                        The controller manager.
ii  ros-melodic-controller-manager-msgs            0.18.3-1bionic.20210505.0249 amd64                        Messages and services for the controller manager.
ii  ros-melodic-costmap-2d                         1.16.7-1bionic.20210505.0359 amd64                        This package provides an implementation of a 2D costmap that takes in sensor data from the world,  [... cut ...]
ii  ros-melodic-webkit-dependency                  1.1.0-0bionic.20200801.04053 amd64                        This encapsulates the WebKit dependency for a specific ROS distribution and its Qt version
ii  ros-melodic-xacro                              1.13.11-1bionic.20210601.153 amd64                        Xacro (XML Macros) Xacro is an XML macro language.
ii  ros-melodic-xmlrpcpp                           1.14.11-1bionic.20210427.131 amd64                        XmlRpc++ is a C++ implementation of the XML-RPC protocol.

```

dgarbuzov, 2021-07-07:
Looks like we're using all the same ROS packages. It looks like you're using an updated version of cmake (3.20) instead of the one packaged for bionic/melodic (3.10). Have you tried using the default cmake? Other than that, maybe the output of `catkin build -v` will show something.

kairenw, 2021-08-24:
Closing out due to no response. Feel free to re-open the ticket if you're still having issues.

closed 2021-08-24 by kairenw

arjunskumar, 2022-02-28:
@dgarbuzov Compile the package with gcc-8, it will solve the issues.

nickle-fang, 2022-04-08:
You can check your g++ version and GCC version. It is caused by the reason that these versions are not the same.