ouster-lidar/ouster-sdk #263: No data is published to topics despite successful sensor initialization

JordySehn, 2021-06-14:
Hi,

I am working with an Ouster OS1-64 with the v2.0 firmware freshly flashed (two separate units actually). I am connecting them direct to my ubuntu 20.04 machine using a link local connection and am able to successfully ping both the sensor and the IPV4 UDP packet address which was assigned by the link local connection after following the software manual instructions. I have also tried using static IP's, but the problem persists.

Furthermore, I am able to read the diagnostics from the http site and everything appears to be running normally and I can see the udp packets coming in to the correct address on wireshark.

When I run the launch file, after a short time it says the sensor successfully initialized and I get no error messages, but nothing is published to any of the ROS topics, nor am I able to visualize the point cloud with the viz node.

Please let me know if you can provide any assistance or if I can provide any additional information.

kairenw, 2021-06-15:
Hi @JordySehn,

It'd help a lot if you could just paste in the terminal output from pinging your sensor, `ifconfig`, and running the launch file. My first guess is that you may not have set the configuration parameter `udp_dest` on your sensor. That should be covered in your user manual.

JordySehn, 2021-06-15:
Hi @Kairenw,

Fortunately I was able to just figure out the problem. My configurations were all correct, but I realized my work's firewall was blocking the lidar and imu udp ports. Manually setting them to 7502 and 7503 respectively in the launch file, and creating an exception in my firewall fixed the problem and everything is functioning as expected.

Thank you anyways though!

closed 2021-06-15 by JordySehn