ouster-lidar/ouster-sdk #264: Range field of Point.h

Betschler, 2021-06-16:
Hi,

I am working with an Ouster OS1-32 Gen 2 as a perceptional Sensor for a positioning system.
According to my understanding, the "Range" field of Point should indicate the measured distance to the point cloud points, as written in the Software User Manual Range is the "range in millimeters, discretized to the nearest 3 millimeters.", right?

So, I was expecting a higher value for measurement points that are far away and a lower one for close ones. 
However, it just seems not to be the case. 
[image]

Am I wrong? If not, any idea, what the problem might be?
Thanks in advance.

kairenw, 2021-06-16:
Hi @Betschler, 

That's correct: the sensor returns range measurements, which should return lower values for closer objects and higher values for farther-away ones. 

I'm having a little trouble interpreting what your screenshot indicates. Perhaps you could tell us what you're looking for and how you generated the cloud?

versatran01, 2021-06-18:
You might want to check if the colormap is inverted in rviz

Betschler, 2021-06-18:
> Hi @Betschler,
>
> That's correct: the sensor returns range measurements, which should return lower values for closer objects and higher values for farther-away ones.
>
> I'm having a little trouble [...]

Hi, I just downloaded the package, fired it up in my room, and watched it with rviz. The firmware of the OS1-32-Beam Gen2 is v2.0.0. 
Somehow the range value seems not to be right. The image shows my room and the stuff inside. The colormap is not inverted and displays the range value by intensity.

The jacket which is very reflective gets a way higher range value than everything around it. It seems to be a bug in the software package since when watching the range value in OusterStudio it looks fine.
 
Here again the image by rviz with the code of this website both .
[image]
Here in OusterStudio:
[image]

Best regards,
Daniel

kairenw, 2021-06-21:
@Betschler -- Oh I see! Thank you for the clarifications.

I believe we fixed this bug in PR #239. Are you on the latest version?

closed 2021-07-07 by dgarbuzov