ouster-lidar/ouster-sdk #265: PCAP files recorded with Ouster Python SDK can not be viewed in Ouster Studio

Radanz, 2021-06-18:
Hi, As the title says: 
I can record scan data into PCAP files either with the client.py example code or my own code and can then view the data fine via python code
However those PCAP files will not load into Ouster Studio, checking the python console in the app shows the following error message:

----
>>> ERROR: In C:\S\4650\LVCore\LidarPlugin\IO\Lidar\Common\vtkLidarReader.cxx, line 301
vtkLidarReader (0000022FD6C05530): Failed to open packet file: C:/Temp/Tests/OS-1-128_122110000381_1024x10_20210618_124324.pcap!
Unknown link type in pcap file. Cannot tell where the payload is.

ERROR: In C:\S\4650\LVCore\LidarPlugin\IO\Lidar\Common\vtkLidarReader.cxx, line 75
vtkLidarReader (0000022FD6C05530): Could not open the pcap file

ERROR: In C:\S\4650\LVCore\LidarPlugin\IO\Lidar\Common\vtkLidarReader.cxx, line 301
vtkLidarReader (0000022FD6C05530): Failed to open packet file: C:/Temp/Tests/OS-1-128_122110000381_1024x10_20210618_124324.pcap!
Unknown link type in pcap file. Cannot tell where the payload is.

ERROR: In C:\S\4650\LVCore\LidarPlugin\IO\Lidar\Common\vtkLidarReader.cxx, line 75
vtkLidarReader (0000022FD6C05530): Could not open the pcap file
-----

Any ideas?
Seems to have a header that is two bytes shorter than a PCAP file recorded by ouster studio itself.
Actual lidar packet data is valid. In the PCAP file from the python sdk the first lidar packet starts on byte 0x52 but from the studio app it starts on 0x54
Does the Python SDK need some special parameter/function to write a valid header?
Thanks.

dgarbuzov, 2021-06-30:
Looks like OusterStudio expects ethernet encapsulation in the pcap files. The pcaps written with the libraries here are using [SLL](https://wiki.wireshark.org/SLL), I think to indicate that there isn't any actual link-layer information being captured from a network interface.

Adding fake ethernet headers doesn't make a lot of sense to me, but I think we'll have to update the defaults to make sure these tools are compatible.

Radanz, 2021-06-30:
I've been communicating with tech support at Ouster via emails and they have tole me that it's a fault with Ouster Studio that needs to be fixed.
As you say Ouster Studio is expecting Ethernet encapsulation and not SLL so it assumes no data.
Since Ouster Studio is developed via a 3rd party then they need to get them to fix it.
I agree faking headers makes no sense, hopefully the studio app devs can patch it soon.

I'm ok to mark this as closed since this lib is fine, and wait for the app to be fixed.
I might suggest a note added to the docs on the encapsulation method to help others understand it.
Thanks.

dgarbuzov, 2021-06-30:
The development cycle on Ouster Studio could be a bit on the longer side, but I think it might be relatively quick to add an option to change the encapsulation in the library here. 

Can you tell us a bit about your use case? We've had some users that record pcaps in Ouster Studio and then want to write some scripts to process them, but not the reverse.

Radanz, 2021-06-30:
I'm creating a mobile lidar data logger that records to ssd for offline post processing.
For myself I just wanted to use the Studio to verify the recorded files visually and play them back.
I'm not doing the post processing myself and since that's still in development I just wanted a quick way to verify my side without having to develop my own solution.

dgarbuzov, 2021-07-01:
Ah -- if you just need to verify that the data was recorded, we've recently added [an example using open3d that you should be able to run directly on a pcap](https://static.ouster.dev/sdk-docs/examples.html#visualization-with-open3d). Hope that helps!

Radanz, 2021-07-02:
Thanks, will give that a go.
Only downside is having to roll back to python 3.8 as Open3D does not support 3.9 as yet.

dgarbuzov, 2021-08-03:
Should be fixed by #275. Updated the default encapsulation to be compatible with what Ouster Studio expects

closed 2021-08-03 by dgarbuzov