ouster-lidar/ouster-sdk #266: "Must specify path for metadata output" even though running it with the sensor

nubertj, 2021-06-19:
Hello,
With the new version of the driver I tried running ouster_ros on our sensor without providing a metadata path.
Nevertheless, it still dies with the following error:
`process[os_node-2]: started with pid [380581]                                                                                            
process[os_cloud_node-3]: started with pid [380582]                                                                                      
process[relay_lidar-4]: started with pid [380587]                                                                                        
[ERROR] [1624116021.770242398]: Must specify path for metadata output                                                                    
process[relay_imu-5]: started with pid [380589]                                                                                          
[ INFO] [1624116021.778133096]: waitForService: Service [/os_node/os_config] has not been advertised, waiting...                         
================================================================================REQUIRED process [os_node-2] has died!                   
process has died [pid 380581, exit code 1, cmd /home/nubertj/rsl_workspaces_2021_06/devel/lib/ouster_ros/os_node __name:=os_node __log:=/
home/nubertj/.ros/log/dc67c94a-d111-11eb-a101-c94ee2210ba7/os_node-2.log].                                                               
log file: /home/nubertj/.ros/log/dc67c94a-d111-11eb-a101-c94ee2210ba7/os_node-2*.log                                                     
Initiating shutdown!                                                                                                                     
================================================================================ 

When running the launch file with the old metadata I recorded 4 weeks ago, everything works fine. I also went back to the last version, where it works on the sensor even without providing the metadata. Then it just generates it.

I am on Ubuntu 20.04.
Thanks a lot!

kairenw, 2021-06-21:
Hi @nubertj,

This is desired behavior on the latest code. The path it is asking you to provide is for the metadata _output_, not input. That is, it will still pull the metadata from the connected sensor, as you can verify [here](https://github.com/ouster-lidar/ouster_example/blob/b25f67315f8d37da36181106ca2bb97aee26c057/ouster_ros/src/os_node.cpp#L210), but it now requires the user to specify a path for output instead of defaulting to `$ROS_HOME`.

Hope that helps!

kairenw, 2021-06-21:
Closing issue. Please feel free to reopen if it isn't resolved.

closed 2021-06-21 by kairenw

nubertj, 2021-06-22:
Hi @kairenw ,
Thanks for your fast reply and your help!
I think it would also be necessary to update the launch file then, there it still says:

> <arg name="metadata" default="" doc="override default metadata file for replays"/>

And in general it would be good if updates would provide some compatibility without breaking the driver after pulling the latest version from GitHub.

Best,
Julian