ouster-lidar/ouster-sdk #267: python open3d viz crashes after a minute on Intel Mac

dllu, 2021-06-22:
Tried running

```
pip install ouster-sdk open3d
python -m ouster.sdk.examples.open3d --pcap ~/os-992114000702.pcap --meta ~/os-992114000702.local.json
```

It works fine for about a minute and then reliably crashes every time.

Relevant logs:

```
Process:               python3.8 [65437]
Path:                  /Users/USER/*/python
Identifier:            python3.8
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        zsh [64574]
Responsible:           Terminal [51019]
User ID:               501

Date/Time:             2021-06-21 16:58:40.439 -0700
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     5.4 (18P4663)
Anonymous UUID:        C3733766-E0EF-4D38-8F1C-B87319FB0EC5

Sleep/Wake UUID:       640D8A55-8EEB-4D8E-9B0C-587D2986B7EF

Time Awake Since Boot: 150000 seconds
Time Since Wake:       26000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [65437]

VM Regions Near 0:
--> 
    __TEXT                      10272a000-102976000    [ 2352K] r-x/r-x SM=COW  /Users/*/*.8

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	000000000000000000 0 + 0
1   pybind.cpython-38-darwin.so   	0x0000000119666812 open3d::visualization::glsl::TextureSimpleShader::UnbindGeometry() + 210
2   pybind.cpython-38-darwin.so   	0x000000011966672e open3d::visualization::glsl::TextureSimpleShader::Release() + 14
3   pybind.cpython-38-darwin.so   	0x00000001196a037d open3d::visualization::glsl::TextureSimpleShader::~TextureSimpleShader() + 29
4   pybind.cpython-38-darwin.so   	0x00000001196a0552 open3d::visualization::glsl::TriangleMeshRenderer::~TriangleMeshRenderer() + 242
5   pybind.cpython-38-darwin.so   	0x00000001196999c8 open3d::visualization::Visualizer::~Visualizer() + 536
6   pybind.cpython-38-darwin.so   	0x00000001196b28c1 open3d::visualization::VisualizerWithKeyCallback::~VisualizerWithKeyCallback() + 65
7   pybind.cpython-38-darwin.so   	0x000000011916104a pybind11::class_<open3d::visualization::VisualizerWithKeyCallback, open3d::visualization::PyVisualizer<open3d::visualization::VisualizerWithKeyCallback>, std::__1::shared_ptr<open3d::visualization::VisualizerWithKeyCallback> >::dealloc(pybind11::detail::value_and_holder&) + 90
8   pybind.cpython-38-darwin.so   	0x0000000118d9baeb pybind11::detail::clear_instance(_object*) + 411
9   pybind.cpython-38-darwin.so   	0x0000000118d946df pybind11_object_dealloc + 15
10  pybind.cpython-38-darwin.so   	0x0000000118d9bd32 pybind11::detail::clear_patients(_object*) + 306
11  pybind.cpython-38-darwin.so   	0x0000000118d9bbac pybind11::detail::clear_instance(_object*) + 604
12  pybind.cpython-38-darwin.so   	0x0000000118d946df pybind11_object_dealloc + 15
13  python                        	0x00000001027652f2 frame_dealloc + 178
14  python                        	0x0000000102881b1e tb_dealloc + 126
15  python                        	0x0000000102881b25 tb_dealloc + 133
16  python                        	0x0000000102825c3d _PyEval_EvalFrameDefault + 12157
17  python                        	0x000000010274ce9d function_code_fastcall + 237
18  python                        	0x000000010282dca2 call_function + 354
19  python                        	0x000000010282a410 _PyEval_EvalFrameDefault + 30544
20  python                        	0x000000010282ed14 _PyEval_EvalCodeWithName + 3268
21  python                        	0x0000000102822c20 PyEval_EvalCode + 48
22  python                        	0x000000010282004a builtin_exec + 618
23  python                        	0x000000010278c631 cfunction_vectorcall_FASTCALL + 177
24  python                        	0x000000010282dca2 call_function + 354
25  python                        	0x000000010282a410 _PyEval_EvalFrameDefault + 30544
26  python                        	0x000000010282ed14 _PyEval_EvalCodeWithName + 3268
27  python                        	0x000000010274d01d _PyFunction_Vectorcall + 253
28  python                        	0x000000010282dca2 call_function + 354
29  python                        	0x000000010282a410 _PyEval_EvalFrameDefault + 30544
30  python                        	0x000000010282ed14 _PyEval_EvalCodeWithName + 3268
31  python                        	0x000000010274d01d _PyFunction_Vectorcall + 253
32  python                        	0x000000010274c834 PyVectorcall_Call + 100
33  python                        	0x0000000102893160 pymain_run_module + 192
34  python                        	0x0000000102892848 Py_RunMain + 1416
35  python                        	0x0000000102893033 pymain_main + 403
36  python                        	0x000000010289308b Py_BytesMain + 43
37  libdyld.dylib                 	0x00007fff203aef5d start + 1
```

dgarbuzov, 2021-06-22:
Thanks for the detailed report. Will try to see if we can reproduce this.

dgarbuzov, 2021-08-03:
Should be fixed on master now

closed 2021-08-03 by dgarbuzov