ouster-lidar/ouster-sdk #271: record pcap with ouster_pcap

mpkuse, 2021-07-01:
Hi, I have modified your ros driver ouster_ros to write the packets to .pcap files. Particularly I modified your 
connection_loop function https://github.com/ouster-lidar/ouster_example/blob/b8b23c35d7b719d69341a438d386a801688aa6a4/ouster_ros/src/os_node.cpp#L77 . 

I also modified the cmake files as needed to compile the os_pcap.cpp. The modified ros node 
call write a pcap file. However as I open it with ouster studio, it does crash with the following message:
```
ERROR: In /home/gitlab-runner/builds/BGDjswhX/0/LidarView/ouster-studio/LVCore/LidarPlugin/IO/Lidar/Common/vtkLidarReader.cxx, line 295
vtkLidarReader (0x613cdb0): Failed to open packet file: /home/manohar/Downloads/ouster.pcap!
Unknown link type in pcap file. Cannot tell where the payload is.
```

Recording from ouster studio is not an option for me because the lidar is on a remote computer on the robot. 
My question is, how can I make this pcap compatible with ouster studio? 

```c++
int connection_loop(ros::NodeHandle& nh, sensor::client& cli,
                    const sensor::sensor_info& info) {
    auto lidar_packet_pub = nh.advertise<PacketMsg>("lidar_packets", 1280);
    auto imu_packet_pub = nh.advertise<PacketMsg>("imu_packets", 100);

    auto pf = sensor::get_format(info);

    PacketMsg lidar_packet, imu_packet;
    lidar_packet.buf.resize(pf.lidar_packet_size + 1);
    imu_packet.buf.resize(pf.imu_packet_size + 1);

    #if 1 // write pcap files
    std::string pcap_fname = "/tmp/ouster.pcap";
    std::cout << "Open a pcap recording file: " << pcap_fname << std::endl;
    auto pcap_record_handle = ouster::sensor_utils::record_initialize( pcap_fname, "192.168.2.208", "192.168.2.109", 1280 );
    #endif // write pcap

    while (ros::ok()) {
        auto state = sensor::poll_client(cli);
        if (state == sensor::EXIT) {
            ROS_INFO("poll_client: caught signal, exiting");
            return EXIT_SUCCESS;
        }
        if (state & sensor::CLIENT_ERROR) {
            ROS_ERROR("poll_client: returned error");
            return EXIT_FAILURE;
        }
        if (state & sensor::LIDAR_DATA) {
            if (sensor::read_lidar_packet(cli, lidar_packet.buf.data(), pf)) {
                lidar_packet_pub.publish(lidar_packet);

                #if 1 // write pcap files

                // for (int icol = 0; icol < pf.columns_per_packet; icol++) {
                    const uint8_t* col_buf = pf.nth_col(0, lidar_packet.buf.data());
                    // std::cout << col_buf[0] << "\t";
                    const std::chrono::nanoseconds ts(pf.col_timestamp(col_buf));

                std::cout << "pf.lidar_packet_size: " << pf.lidar_packet_size << "\t";
                std::cout << "pf.columns_per_packet: " << pf.columns_per_packet << "\t";
                std::cout << "(uint64_t) ts.count() / 1000 : " << (uint64_t) ts.count() / 1000  << "\t";
                std::cout << std::endl;

                ouster::sensor_utils::record_packet( *pcap_record_handle,
                    7502, 7502,
                    lidar_packet.buf.data(), pf.lidar_packet_size ,
                    (uint64_t) ts.count() / 1000
                );
                #endif
            }
        }
        if (state & sensor::IMU_DATA) {
            if (sensor::read_imu_packet(cli, imu_packet.buf.data(), pf))
                imu_packet_pub.publish(imu_packet);
        }
        ros::spinOnce();
    }

    #if 1 // write pcap files
    std::cout << "Done writing a pcap recording file: " << pcap_fname << std::endl;
    ouster::sensor_utils::record_uninitialize( *pcap_record_handle );
    #endif

    return EXIT_SUCCESS;
}
```

dgarbuzov, 2021-07-01:
Believe this is the same issue as #265. We'll have a workaround soon

gbull25, 2021-07-14:
Hello @mpkuse!

Could you please share modified cmake files for this case? This would be a great help.

Thank you in advance.

dgarbuzov, 2021-08-03:
Should be fixed by #275. Updated the default encapsulation to be compatible with what Ouster Studio expects

closed 2021-08-03 by dgarbuzov