ouster-lidar/ouster-sdk #273: running ouster_client_example

KatuunuXVI, 2021-07-23:
I'm currently working with a NVIDIA Jetson and I have it connected to a lidar sensor via ethernet cord (through a upoe switch) when I run ipconfig, the given ip address from eth0 is the following,

```
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 169.254.1.250  netmask 255.255.0.0  broadcast 169.254.255.255
        ether 48:b0:2d:15:da:cb  txqueuelen 1000  (Ethernet)
        RX packets 6130323  bytes 7015456418 (7.0 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1227997  bytes 67763352 (67.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 37
```

However when running the example code, as follows (169.168.0.208 is the ip address of the Jetson),
`./ouster_client_example 169.254.1.250 169.168.0.208`
I get a connection failure. Is there a method to verify that a connection to the lidar sensor from a linux environment is possible, or a method to verify the IP address of the sensor?

closed 2021-07-28 by KatuunuXVI

pulver22, 2021-11-04:
I actually have the same error while connecting the LIDAR to a ZOTAC.

arunumd, 2023-05-26:
You can try `./client_example 169.254.1.250 ""`