ouster-lidar/ouster-sdk #274: Calling config service failed with OS1 in Ouster Ros

lid747, 2021-07-27:
Hello,

I'm getting this error when executing ouster.launch from the ouster_example packages.

[ INFO] [1627429850.959234192]: Client version: 0.2.0-dev0+2b49e6a-dirty-release
[ INFO] [1627429850.961517125]: Connecting to 192.168.0.109; sending data to 192.168.0.110
[ INFO] [1627429850.961649898]: Waiting for sensor to initialize
process[relay_imu-5]: started with pid [10948]
[ERROR] [1627429871.306619447]: Failed to initialize sensor at: 192.168.0.109
[ERROR] [1627429871.313654078]: Calling config service failed

================================================================================REQUIRED process [os_node-2] has died!

process has died [pid 10939, exit code 1, cmd /home/ldr747_brund/BrunD_Project/brund_ws/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/ldr747_brund/.ros/log/77f97962-ef35-11eb-bc56-00e04c68d221/os_node-2.log].
log file: /home/ldr747_brund/.ros/log/77f97962-ef35-11eb-bc56-00e04c68d221/os_node-2*.log
Initiating shutdown!

================================================================================

[relay_imu-5] killing on exit
[relay_lidar-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor
shutting down processing monitor complete
done

From Ouster I'm getting:

{
  "alerts": {
    "active": [
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 48,
        "id": "0x01000015",
        "level": "WARNING",
        "msg": "Client not reachable on lidar data port; check udp_ip, udp_port_lidar, and client machine.",
        "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.0.110:24001",
        "realtime": "1627429871309996800"
      },
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 49,
        "id": "0x01000018",
        "level": "WARNING",
        "msg": "Client not reachable on IMU port; check udp_ip, udp_port_imu, and client machine.",
        "msg_verbose": "Failed to send imu UDP data to destination host 192.168.0.110:24002",
        "realtime": "1627429871319029504"
      }
    ],
    "log": [
      {
        "active": false,
        "category": "UDP_TRANSMISSION",
        "cursor": 18,
        "id": "0x01000015",
        "level": "WARNING",
        "msg": "Client not reachable on lidar data port; check udp_ip, udp_port_lidar, and client machine.",
        "msg_verbose": "Cleared by reinitialization.",
        "realtime": "1627427764050118400"
      },
      {
        "active": false,
        "category": "UDP_TRANSMISSION",
        "cursor": 19,
        "id": "0x01000018",
        "level": "WARNING",
        "msg": "Client not reachable on IMU port; check udp_ip, udp_port_imu, and client machine.",
        "msg_verbose": "Cleared by reinitialization.",
        "realtime": "1627427764050431744"

kairenw, 2021-08-16:
Hi @lid747, 

Sorry for the long wait. Are you still having issues connecting to your sensor? I would double check that your sensor IP and your destination IP are set correctly? If you upgrade to FW 2.1, you can check the destination IP a little more easily using the Beta Configuration interface.

kairenw, 2022-02-03:
Closing due to lack of reply. Feel free to reopen if this is not addressed!

closed 2022-02-03 by kairenw