ouster-lidar/ouster-sdk #276: Convert recorded rosbag file to numpy array

baheytharwat, 2021-08-03:
I have a recorded rosbag file for OS1-64 lidar and tried to read the point cloud of each message through the topic `/os1_cloud_node/points` and convert it into a numpy array using different libraries such as pcl and pypcd but I got a corrupted point cloud. Although I visualized the rosbag file using RVIZ and it seems OK. 
Do you provide any tutorials or scripts that can help me converting a message(PointCloud2) from `/os1_cloud_node/points` topic to a numpy array that contains x,y,z and intensity?

closed 2021-08-08 by baheytharwat

hujunhan, 2023-02-07:
I have a similar problem, did you solve it?

kairenw, 2023-02-07:
Hi @hujunhan can you start a new issue for this, thanks!