ouster-lidar/ouster-sdk #277: ROS driver multicast support

ShepelIlya, 2021-08-03:
I have multiple Ouster OS1-32D and OS0-64D devices. I want to receive multicast data from them on several computers, and publish point clouds using the ROS driver. Does the driver currently support multicast traffic?

I have configured the udp_dest parameter on my lidars, and I can see multicast traffic through the wireshark, but the driver does not publish messages. Are there any instructions for running the ROS driver via multicast messages?

ShepelIlya, 2021-08-04:
I added multicast support in #278.

yt, 2021-08-17:
Can also confirm multicast doesn't work on an OS1-32D device. 
Tested with joining to a multicast group `ip addr add 239.1.0.1 dev myINT autojoin` and giving the broadcast ip to udp_dest

ShepelIlya, 2021-08-17:
@yt can you publish your launch file that doesn't work?
if you fill arg `mtp_dest` in launch file with ip addr of you host interface and `udp_dest` with you multicast group it will work,.
For example:
launch on computer 1 (this launch reconfigures ouster at start):
```xml
<!-- Central Ouster OS0-64D -->
<include file="$(find ouster_ros)/ouster.launch" >
    <arg name="sensor_hostname" value="192.168.1.34" /> <!-- ip addr of sensor -->
    <arg name="udp_dest" value="239.168.1.34" /> <!-- multicast group of sensor -->
    <arg name="mtp_dest" default="192.168.1.110" /> <!-- ip addr of host which network iface will be added to multicast group automatically at driver start -->
    <arg name="lidar_port" value="7502" />
    <arg name="imu_port" value="7503" />
    <arg name="replay" value="false" />
    <arg name="sensor_init" default="false" /> <!-- if this flag is false then reconfigure sensor at start -->
    <arg name="lidar_mode" value="2048x10" />
    <arg name="timestamp_mode" value="TIME_FROM_PTP_1588" />
    <arg name="metadata" value="metadata_os0-64D.json"/> <!-- copy this to another multicast clients -->
    <arg name="viz" value="false" doc="whether to run a simple visualizer"/>
    <arg name="tf_prefix" value="os64central" doc="namespace for tf transforms"/>
    <arg name="publish_transform" value="false" doc="Pusblish transforms from sensor to imu and lidar"/>
</include>
```

launch on computer 2:
```xml
<!-- Central Ouster OS0-64D -->
<include file="$(find ouster_ros)/ouster.launch" >
    <arg name="sensor_hostname" value="192.168.1.34" /> <!-- ip addr of sensor -->
    <arg name="udp_dest" value="239.168.1.34" /> <!-- multicast group of sensor -->
    <arg name="mtp_dest" default="192.168.1.111" /> <!-- ip addr of host which network iface will be added to multicast group automatically at driver start -->
    <arg name="lidar_port" value="7502" /> <!-- same as on computer 1 -->
    <arg name="imu_port" value="7503" /> <!-- same as on computer 1 -->
    <arg name="replay" value="false" />
    <arg name="sensor_init" default="true" /> <!-- dont reinit the sensor config and get params from metafile -->
    <arg name="lidar_mode" value="2048x10" />
    <arg name="timestamp_mode" value="TIME_FROM_PTP_1588" />
    <arg name="metadata" value="metadata_os0-64D.json" /> <!-- same file as on computer 1 -->
    <arg name="viz" value="false" doc="whether to run a simple visualizer"/>
    <arg name="tf_prefix" value="os64central" doc="namespace for tf transforms"/>
    <arg name="publish_transform" value="false" doc="Pusblish transforms from sensor to imu and lidar"/>
</include>
```