ouster-lidar/ouster-sdk #279: ouster_ros package not compiled when compiling with ros

Fingolfin007, 2021-08-05:
I compiled the ros driver with the following procedure on a Ubuntu 20 with ros noetic:

- git clone the repo to <some-dir>
- create symlink of <some-dir> under xx-ws/src/
- colcon build (in the xx-ws)

What happened is that the ouster_example repo got built, but the ouster_ros package never got built properly. After sourcing the setup.bash file, the ros package still cannot be recognized. 

I checked the xx-ws/install/ folder, there's no ouster_ros, but only ouster_example, which contains the compiled binaries for ./ouster_client_example and ./simple_viz. 

I tried to only create symlnk for ouster_ros/ instead of ouster_example, but that resulted in it's own errors saying some files cannot be reached. 

Any solution possible?

AchmadFathoni, 2021-08-05:
Have you tried to build it with catkin_make?

ShepelIlya, 2021-08-05:
There is no code for ROS2 now. You can try to create your own ROS2 node for ouster based on this repo.

Fingolfin007, 2021-08-05:
I am using colcon build for ROS 1. I tried catkin_make as per suggestion by @AchmadFathoni , and it did build. So I dived into the cmakelists try to make it work for colcon build as well. I found out that if I go into the top level cmakelists in the repo, and change the line add_subdirectory(ouster_client) into add_subdirectory(ouster_ros) it works. 

for those who might be curious of why this worked: this line allows colcon build to find the subdirectory ouster_ros. Whereas removing the add_subdirectory(ouster_client) is necessary because ouster_ros already depends on ouster_client, adding the sub_directory in the top level cmakelists just creates duplication and results in failure.

closed 2021-08-05 by Fingolfin007