ouster-lidar/ouster-sdk #280: Ouster timestamp vs ROS wall time configuration

tahir1069, 2021-08-10:
I have integrated an Ouster OS1 lidar to Clearpath Husky and configured everything on Husky and things seems to work fine. But I get an error:

```
[ WARN] [1628603138.778912200]: MessageFilter [target=odom ]: Dropped 100.00% of messages so far. Please turn the [ros.gmapping.message_filter] rosconsole logger to DEBUG for more information.
[ WARN] [1628603138.780674588]: MessageFilter [target=odom ]:   The majority of dropped messages were due to messages growing older than the TF cache time.  The last message's timestamp was: 2128.263543, and the last frame_id was: os_sensor

```

The main issue here is that the ouster ros driver is using internal clock and ros is using system time. I tried to set the parameter `timestamp_mode` to `TIME_FROM_PTP_1588`. This seems to update and I can see that in Ouster diagnostics. 

How ever I still get the same error after setting this up. I have tried to follow #74(Also this seems to be valid for previous version as in the current version this parameter I think can be set up from the web gui under Configuration) but I think my system doesn't support that ptp setup stating the error:
```
ptp4l[3330.313]: interface 'br0' does not support requested timestamping mode
failed to create a clock
```
A small ros discussion can be found [here](https://answers.ros.org/question/384054/messages-dropped-in-tf-while-running-a-gmapping-node/).

Any thoughts on that will highly be appreciated.

Samahu, 2022-08-20:
Hi @tahir1069, if you are having trouble configuring the ptp clock synchronization properly on your system, I can refer you to [our support team](https://ouster.atlassian.net/servicedesk/customer/portal/8)  and they will be able to assist you with this.

With regard to feature described in https://github.com/ouster-lidar/ouster_example/issues/74, this has been implemented along with plenty of other improvements to the ROS driver, please watch for the current pending release.

Best

Samahu, 2022-08-24:
A new release has been published which implements the feature discussed in #74. Please refer to https://github.com/ouster-lidar/ouster_example/pull/420 and https://github.com/ouster-lidar/ouster_example/pull/421 for details

closed 2022-08-24 by Samahu