ouster-lidar/ouster-sdk #283: .pcap reader in matlab

pavankumarbn, 2021-08-13:
Hello All,

I have ouster recorded .pcap data. How can I read this data using matlab? Please help me out. 

Thanks,
Pavan

kairenw, 2021-08-13:
Hi @pavankumarbn,

Could you let us know a little bit of your use case? You may find it easiest to read your recorded .pcap using the [Ouster Python SDK](https://static.ouster.dev/sdk-docs/quickstart.html), and save the loaded data from numpy into `.mat` or .csv to port into matlab.

pavankumarbn, 2021-08-14:
Hi @kairenw, Many thanks for your response. Sure I will try it out as you suggested. We are working on object detection and tracking using ouster data on matlab platform. Regarding conversion, please let me know if there is any demo example on this ?

kairenw, 2021-08-16:
Hey @pavankumarbn,

We don't have a demo example, but it should be fairly straightforward using something like scipy's [savemat](https://docs.scipy.org/doc/scipy/reference/generated/scipy.io.savemat.html). You can try working off our save to csv [example](https://static.ouster.dev/sdk-docs/examples.html#converting-pcaps-to-csv). Let us know how it goes!

pavankumarbn, 2021-08-17:
Hi @kairenw, Thank you for your response. It is much appreciated. I followed as you suggested and worked on the CSV example. When I run `python3 -m ouster.sdk.examples.pcap OS1_128.pcap OS1_2048x10_128.json pcap-to-csv --scan-num 5` ; it is working fine. However, when I try to run the complete source code as given it is throwing some errors. I will post the complete code in another open issue. Please help me with that.

kairenw, 2021-08-17:
Great! I'm going to close this out and keep working with you on the other ticket. Thanks!

closed 2021-08-17 by kairenw