ouster-lidar/ouster-sdk #284: NameError: name 'field_names' is not defined | .pcap to .csv

pavankumarbn, 2021-08-15:
Hello All,

I am trying to convert ouster recorded .pcap to data to CSV format. When I try to run the linked code https://static.ouster.dev/sdk-docs/api.html#ouster.sdk.examples.pcap.pcap_to_csv  it throwing  ` header = '\n'.join([f'frame num: {idx}', field_names])
NameError: name 'field_names' is not defined`. Please help me with how to fix this issue? 

Thanks,
Pavan

kairenw, 2021-08-16:
Hi @pavankumarbn,

Could you paste in the terminal logs of how you're running it, and the subsequent error? Thanks!

pavankumarbn, 2021-08-17:
Hi, @kairenw, Thanks for your response. Please find the complete code below: 
`import os
import numpy as np
from ouster import client
from ouster import pcap

from itertools import islice
pcap_path = 'sample.pcap'
metadata_path = 'sample.json'
csv_dir = 'test_csv/'
csv_base = '/test'
csv_ext = '.gz'
 # precompute xyzlut to save computation in a loop
 
with open(metadata_path, 'r') as f:
    metadata = client.SensorInfo(f.read())
source = pcap.Pcap(pcap_path, metadata) 
xyzlut = client.XYZLut(metadata)
 
# create an iterator of LidarScans from pcap and bound it if num is specified
scans = iter(client.Scans(source))
num = 5
if num:
    scans = islice(scans, num)

for idx, scan in enumerate(scans):

   # copy per-column timestamps for each channel
   col_timestamps = scan.header(client.ColHeader.TIMESTAMP)
   timestamps = np.tile(col_timestamps, (scan.h, 1))

   # grab channel data
   fields_values = [scan.field(ch) for ch in client.ChanField]
   # use integer mm to avoid loss of precision casting timestamps
   xyz = (xyzlut(scan) * 1000).astype(np.int64)
   # get all data as one H x W x 8 int64 array for savetxt()
   frame = np.dstack((timestamps, *fields_values, xyz))
   # not necessary, but output points in "image" vs. staggered order
   frame = client.destagger(metadata, frame)
   # write csv out to file
   csv_path = os.path.join(csv_dir, f'{csv_base}_{idx:06d}.{csv_ext}')
   print(f'write frame #{idx}, to file: {csv_path}')
   header = '\n'.join([f'frame num: {idx}', field_names])
   np.savetxt(csv_path,frame.reshape(-1, frame.shape[2]),fmt=field_fmts,delimiter=',',header=header)`

kairenw, 2021-08-17:
Hey @pavankumarbn,

I think you just need to define `field_names` and `field_fmts`. You probably just want these two lines from the [pcap-to-csv example](https://github.com/ouster-lidar/ouster_example/blob/master/python/src/ouster/sdk/examples/pcap.py#L191):

`field_names` = 'TIMESTAMP (ns), RANGE (mm), SIGNAL, NEAR_IR, REFLECTIVITY, X (mm), Y (mm), Z (mm)'
 field_fmts = ['%d', '%d', '%d', '%d', '%d', '%d', '%d', '%d']`

I think the indentations got lost when you pasted in your code, but you probably want to do it up top before any loops.

pavankumarbn, 2021-08-17:
Hi @kairenw, Thanks a lot for your kind help and support. I added these two lines, it is working perfectly fine now.  I have another question, is it possible to convert the ouster recorded .pcap file .bag (ROS format). If yes, please guide me on how to do it?

kairenw, 2021-08-17:
Hi @pavankumarbn, 

Glad that worked out for you. 

We don't currently have code that converts the Ouster record .pcap file to .bag. Since this is a duplicate issue to #172, I will close this issue now.

Thanks!

pavankumarbn, 2021-08-17:
Hi @kairenw, Sure I will go through it. Once again many thanks for your timely help.