ouster-lidar/ouster-sdk #285: "Unexpected number of beam_altitude_angles" for OS1-128 data

michielvlaminck, 2021-08-18:
I am currently using the python ouster-sdk version 0.2.2 to read and process raw pcap files I recorded using the OS1-128 scanner.

However, when opening them using the python code below, I get the following error: "ValueError: Unexpected number of beam_altitude_angles".

`
with open(config_path, 'r') as f:
        config = client.SensorInfo(f.read())

source = pcap.Pcap(pcap_path, config)
scans = client.Scans(source)`

I generated the config file containing the beam altitude angles using Ouster Studio and the same OS1-128 scanner.

I also noticed that only version 3.6.0-984 of Ouster Studio can read the pcap file. Version 3.6.0-777 of Ouster Studio fails, so perhaps the two issues are linked.

I also have a OS1-64 lidar scanner and for that sensor I can correctly open and process the pcap files using the same code (with another config file of course).

Anyone knows how to solve this issue?

kairenw, 2021-08-18:
Hi @michielvlaminck,

Mind [generating a config file through the Python SDK](https://static.ouster.dev/sdk-docs/examples.html#obtaining-sensor-metadata) as well, and trying it with your recorded data? Let us know if that works or still fails for you! 

Also, which FW version is on each of your sensors?

michielvlaminck, 2021-08-19:
Hi @kairenw,

Thanks for your swift reply! What difference does it make to generate a config file through the Python SDK? I guess the beam altitude angles will be the same so the json files won't be different? Or is there some difference in syntax? In that case I can just manually adjust the json file?

In any case, I will try it next week, when I have access to the scanners.

I remember that for the OS1-64 I have upgraded it to the latest firmware version. The OS1-128 still has the firmware from when it was shipped (in Spring 2020).

Best regards,
Michiel

kairenw, 2021-08-19:
Hi @michielvlaminck,

Theoretically it shouldn't make a difference. The beam altitude and azimuth angles should be more or less fixed from your sensor, although because other things about your sensor configuration will change, it's better to take down a new metadata every time you record data. By asking you to use the Python SDK to generate a new one, I was just hoping you could generate a functional config file, and we could start from there to figure out what happened to the other.

It's not recommended to manually adjust a json file, as the beam angles differ by sensor, so your resulting xyz calculations will be off. Because of that, I'd rather not talk about how to hack a jscon file to make it work on our Github. If you'd like to work on this before next week by editing the json manually, you could send a ticket to our [technical support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86). Mention this ticket and that I asked you to reach me through that portal in the message, and we can pick this up over email. 

[FW 2.1 was released in June](https://ouster.com/blog/firmware-2.1.1-better-perception-performance-with-improved-reflectivity-and-signal-multiplier-mode/) and has some nice features -- I encourage you to update the firmware when you get a chance!

kairenw, 2021-08-20:
Closing issue as we are working with the customer behind the scenes.

For anyone who has found this issue by looking through the backlog and thinks they have a similar issue, please [update your Ouster Studio to the latest version.](https://ouster.com/downloads/) before generating metadata files with it. Or you can use this [metadata-generating function in the python SDK](https://static.ouster.dev/sdk-docs/examples.html#obtaining-sensor-metadata).

Thanks!

closed 2021-08-20 by kairenw