ouster-lidar/ouster-sdk #286: [ERROR] [1629349784.202827955]: Must specify path for metadata output

TheAdamP, 2021-08-19:
Hey a similar issue was posted here before but I couldn't see any clarification on how to fix. The error is as follows:
started core service [/rosout]
process[os_node-2]: started with pid [3335]
process[os_cloud_node-3]: started with pid [3336]
process[relay_lidar-4]: started with pid [3341]
[ERROR] [1629349784.202827955]: Must specify path for metadata output
process[relay_imu-5]: started with pid [3347]
[ INFO] [1629349784.212469255]: waitForService: Service [/os_node/os_config] has not been advertised, waiting...
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 3335, exit code 1, cmd /home/adam/myworkspace/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/adam/.ros/log/a9f88548-00ab-11ec-8a9e-9cb6d00b3cfb/os_node-2.log].
log file: /home/adam/.ros/log/a9f88548-00ab-11ec-8a9e-9cb6d00b3cfb/os_node-2*.log
Initiating shutdown!

So I understand this is an issue with the metadata file path however I am specifying a file path when I launch: 
roslaunch ouster_ros ouster.launch sensor_hostname:=os1-991832000783.local \ metadata:=/home/adam/json \ udp_dest:=169.254.217.143 \ lidar_mode:=1024x20 \ rviz:=true

Any help would be appreciated I may have an error elsewhere

kairenw, 2021-08-19:
Hey @TheAdamP,

If you could remove all the `\` from your command, I think it's escaping the space and reading it as a normal character and therefore feeding in the argument " metadata" instead of `metadata` to ros, if that makes sense. You can also remove the space between `\` and `metadata`, to make `\metadata`, etc, but the `\` is optional! 

Also: to show the viz the option is `viz:=true` not `rviz:=...` 

Hope that helps!

TheAdamP, 2021-08-19:
Hey thanks for the quick response!
That seemed to have helped but I still have an issue with the path:
[ WARN] [1629358835.751421546]: Failed to write metadata to /home/adam/json; check that the path is valid. If you provided a relative path, please note that the working directory of all ROS nodes is set by default to $ROS_HOME
[ERROR] [1629358835.751713189]: Exiting because of failure to write metadata path
[ERROR] [1629358835.755682900]: Calling config service failed
[ERROR] [1629358835.756144265]: Calling os config service failed

I really just want to get the lidar up and running, the path I specified was valid but still crashes. Am I missing something in defining an absolute path? whats the simplest path I can specify that would get it saved to anywhere in my home.

Thanks, 
Adam

kairenw, 2021-08-19:
Not sure what's going on -- if the path is valid, it really shouldn't crash. If you just want to get your lidar up and running with ROS, you can just use a relative path, i.e., `metadata:=metadata.json`. It'll write to `~/.ros/metadat.json`, and you can copy it over later if you like.

TheAdamP, 2021-08-19:
Thank you this worked I will copy it later across later!

kairenw, 2021-08-19:
:tada: Hooray! 

I'm gonna close this ticket out for now. I'd double-check the permissions on the absolute path, possibly. Not sure what else it could be. Let us know if you have any other questions!

closed 2021-08-19 by kairenw