ouster-lidar/ouster-sdk #287: ouster-sdk is incompatable with zeromq

jonbakerfish, 2021-08-19:
Hi, I have a program using zeromq to communicate with other processes. When running ouster-sdk alone, I can capture the lidar data normally but not with zeromq running. The SDK just crash without too much information. Can you help me out?

bexcite, 2021-12-14:
Hi @jonbakerfish,

`ClientTimeout` error means that client haven't received any data packets from Ouster sensor. Common config settings to check in such situation is `udp_dest` and `udp_port_lidar` is set to what is passed to the client. Also some examples may differ in how configuration is performed, which may lead to different result.

I've tried to run ouster client and simple_viz while running ZeroMQ asynsrc example (Chapter3 https://zguide.zeromq.org/docs/chapter3/) in background and seems all processes works fine together.

jonbakerfish, 2021-12-15:
@bexcite Have you tried running ZeroMQ and ouster in a single process but with different threads? The `ClientTimeout` error happens in this case.

dgarbuzov, 2021-12-28:
Think for anyone to help with this one, we'd need a minimal example that reproduces the issue. The only way I can think that the two packages might interfere is if they both try to listen on the same UDP port (and both are using SO_REUSEADDR/SO_REUSEPORT).

closed 2023-08-14 by twslankard