ouster-lidar/ouster-sdk #289: disable IMU package sending

jonbakerfish, 2021-08-22:
Hi, is it possible to configure the OS1-128 lidar to stop sending the IMU packages?

kairenw, 2021-08-24:
Hi @jonbakerfish, 

Care to tell us a little about your use case and why you'd like to stop the IMU packets?

jonbakerfish, 2021-08-25:
Hi, we have other IMU sensors in the system and I assume that stop sending the IMU packages can save the bandwidth & processing time quite a bit.

dgarbuzov, 2021-12-28:
The bandwidth used by IMU data should be negligible. I don't believe there's a supported way to turn it off on the sensor itself, but you can avoid receiving IMU data by intentionally listening on the wrong port (as long as you're not using the `init_client()` overload that configures ports and lidar_mode etc. automatically).

kairenw, 2023-07-07:
Closing - this is a sensor feature request as opposed to a request for the SDK. Please contact an Ouster sales representative or the FAE who supports you to make this feature request.

closed 2023-07-07 by kairenw

themightyoarfish, 2025-04-22:
Regardless of whether the sensor could be prevented from sending the packets, it is a problem that the client cannot be trivially made to not receive them. Usually one would run `poll_client()` in a loop, and that returns if there is lidar data or imu data. Because imu data is so frequent, this causes almost a busy wait on my system at least and uses a lot of CPU just to receive the packets. If I am not interested in the IMU data, is there any way other than deliberately use a wrong imu port? This is not robust if i don't control the entire system, or requires additional logic to find a free port. Seems unnecessarily complex.

themightyoarfish, 2025-04-22:
Actually you cant listen to a wrong imu port, as stated above, if you use `init_client()` so what are my options here, other than fork the SDK?

themightyoarfish, 2025-04-22:
For everyone stumped about their CPU usage when looping `poll_client()`, make sure to `read_lidar_packet()` each time, even if you don't care about the data, otherwise the next call to `poll_client()` will return immediately and you have basically a busy wait on your hands.