ouster-lidar/ouster-sdk #291: ROS driver timestamp not correct

Fingolfin007, 2021-08-23:
In the current driver, the ros point cloud message time stamp is using the chrono stamped time (std::chrono::nanoseconds{0}) at the moment when callback to the lidar packet was triggered. I am making a suggestion to switch to use ros::Time::now() for short term, and use the raw packet information in the long run.

The reasoning is here:

When one uses ouster's ros driver, one usually would assume one of the following to be true for the timestamp of the ros message:
Either
- the time stamp of the point cloud message is populated from the raw ethernet-transported packets (which ensures the accuracy of the timestamp)
Or
- the time stamp of the point cloud message is populated as a stamped time using ros::Time::now() (this way the integration of ouster data becomes seamless for usage in ros for the purpose of tf searching, time stamp diff comparison with other ros messages, etc)

There are of course other more sophisticated ways to deliver both accuracy and integration benefits, but the community would definitely be happy to settle for one of the straight forward ways mentioned above. However, by using chrono (and not stamping the time at the early most moment when the lidar packets arrived), neither accuracy nor convenience was achieved.

Jaetriel, 2021-08-31:
Hello @Fingolfin007 , would this be a huge change in the ouster driver currently? I'd be interested to give it a try to use ros timestamps because I have come across some issues where timestamping works but is inconsistent (sometimes I get higher delays for no apparent reason) or I get some one-off errors, but I haven't checked what changes need to happen in the code for the driver

jsundh, 2021-09-21:
> In the current driver, the ros point cloud message time stamp is using the chrono stamped time (std::chrono::nanoseconds{0}) at the moment when callback to the lidar packet was triggered.

But this is not the case - the timestamps from the measurement blocks in the lidar packets are used.

Look at the code that creates the point cloud message: https://github.com/ouster-lidar/ouster_example/blob/47f25ed29ab0b4b6d32c3209fa1b53ea46751d0c/ouster_ros/src/os_cloud_node.cpp#L63-L73

When the `ScanBatcher` returns true and the `LidarScan` is complete, the `std::find_if` on line 64 returns the first measurement block header where the timestamp _is not zero_. If this header is found, its timestamp is used both to set the point timestamps when creating the PCL PointCloud (line 69) and as the `header.stamp` when creating the ROS PointCloud2 message (line 71).

So the timestamp is accurate and I think this is preferable to using `ros::Time::now()`, because in many cases you want that accuracy.

kairenw, 2021-10-12:
Thanks @jsundh.

Closing issue as won't do, as I think we'd prefer to use the timestamp reading from the sensor over `ros::Time::now()`.