ouster-lidar/ouster-sdk #292: Python sdk

islamtalha01, 2021-08-24:
Hi 
         I am trying to use python SDK with my lidar connected with my laptop through an ethernet interface. I am following a webinar video which is held by ouster org. Following that video, I have created my virtual env and also install python SDK for ouster using pip. Now after defining an environment variable as using the command "export  SENSOR_HOSTNAME=os1-XXX"Where os1-xxx is my lidar hostname, I am using a command for live data streaming that is "python3 -m ouster.SDK.examples.client $SENSOR_HOSTNAME live-plot-signal".By executing this, a traceback appears while it does not appear on the webinar process. I am attaching the screenshot of the traceback. kindly advise me where I am doing wrong.
[image]

kairenw, 2021-08-24:
Hi @islamtalha01,

You probably need to set your udp_dest on the sensor. There are a lot of ways to do that, but probably the easiest way for now, is to make sure you're on FW 2.1, and change it onthe beta web configuration available at your sensor homepage (http://os1-992006000049.local). It's the first field "UDP Destination Address", and if you don't have anything fancy going on with your networking, you can just hit "Set Local", then "Apply Config".

Let us know how it goes!

islamtalha01, 2021-08-25:
Thanks for your help.
I managed to get live stream of data

kairenw, 2021-08-25:
So glad! Closing out this ticket now.

closed 2021-08-25 by kairenw

islamtalha01, 2021-08-25:
Cool, thanks!