ouster-lidar/ouster-sdk #293: catkin_make failed  "invalid static_cast from type ‘Eigen ..."

yinwu33, 2021-08-24:
Hello, I meet an error when I follow the introduction in Readme and did `catkin_make`
The result shows like follows:

`Base path: /home/ubuntu/Data/Dropbox/workspace/test/ouster_example
Source space: /home/ubuntu/Data/Dropbox/workspace/test/ouster_example/src
Build space: /home/ubuntu/Data/Dropbox/workspace/test/ouster_example/build
Devel space: /home/ubuntu/Data/Dropbox/workspace/test/ouster_example/devel
Install space: /home/ubuntu/Data/Dropbox/workspace/test/ouster_example/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/build"
####
####
#### Running command: "make -j16 -l16" in "/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/build"
####
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target generate_build_header
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target _ouster_ros_generate_messages_check_deps_PacketMsg
[  0%] Built target _ouster_ros_generate_messages_check_deps_OSConfigSrv
[ 22%] Built target ouster_client
[ 29%] Built target ouster_ros_generate_messages_lisp
[ 48%] Built target ouster_ros_generate_messages_cpp
[ 48%] Built target ouster_ros_generate_messages_eus
[ 62%] Built target ouster_ros_generate_messages_py
[ 70%] Built target ouster_ros_generate_messages_nodejs
[ 70%] Built target ouster_ros_gencpp
[ 70%] Built target ouster_ros_generate_messages
[ 74%] Building CXX object ouster_example/ouster_ros/CMakeFiles/ouster_ros.dir/src/ros.cpp.o
/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/src/ouster_example/ouster_ros/src/ros.cpp: In function ‘void ouster_ros::scan_to_cloud(const ouster::XYZLut&, ouster::LidarScan::ts_t, const ouster::LidarScan&, ouster_ros::Cloud&)’:
/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/src/ouster_example/ouster_ros/src/ros.cpp:80:69: error: invalid static_cast from type ‘Eigen::internal::enable_if<true, Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field> >::type’ {aka ‘Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field>’} to type ‘float’
   80 |                 static_cast<float>(pix(ouster::LidarScan::INTENSITY)),
      |                                                                     ^
/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/src/ouster_example/ouster_ros/src/ros.cpp:82:75: error: invalid static_cast from type ‘Eigen::internal::enable_if<true, Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field> >::type’ {aka ‘Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field>’} to type ‘uint16_t’ {aka ‘short unsigned int’}
   82 |                 static_cast<uint16_t>(pix(ouster::LidarScan::REFLECTIVITY)),
      |                                                                           ^
/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/src/ouster_example/ouster_ros/src/ros.cpp:84:70: error: invalid static_cast from type ‘Eigen::internal::enable_if<true, Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field> >::type’ {aka ‘Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field>’} to type ‘uint16_t’ {aka ‘short unsigned int’}
   84 |                 static_cast<uint16_t>(pix(ouster::LidarScan::AMBIENT)),
      |                                                                      ^
/home/ubuntu/Data/Dropbox/workspace/test/ouster_example/src/ouster_example/ouster_ros/src/ros.cpp:85:68: error: invalid static_cast from type ‘Eigen::internal::enable_if<true, Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field> >::type’ {aka ‘Eigen::IndexedView<const Eigen::Block<const Eigen::Array<unsigned int, -1, 4>, 1, 4, false>, Eigen::internal::SingleRange, ouster::LidarScan::Field>’} to type ‘uint32_t’ {aka ‘unsigned int’}
   85 |                 static_cast<uint32_t>(pix(ouster::LidarScan::RANGE))};
      |                                                                    ^
make[2]: *** [ouster_example/ouster_ros/CMakeFiles/ouster_ros.dir/build.make:63: ouster_example/ouster_ros/CMakeFiles/ouster_ros.dir/src/ros.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:551: ouster_example/ouster_ros/CMakeFiles/ouster_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed
`

I tried to reinstall `ros-noetic` and `libeigen3-dev` but doesn't work. Does anyone has an idea?

closed 2021-08-24 by yinwu33

kairenw, 2021-08-24:
Hi @yinwu33, 

Was this related to the error you reported [here](https://github.com/ouster-lidar/ouster_example/issues/137#issuecomment-898549256), and was it solved by updating the firmware?

yinwu33, 2021-08-24:
> Hi @yinwu33,
> 
> Was this related to the error you reported [here](https://github.com/ouster-lidar/ouster_example/issues/137#issuecomment-898549256), and was it solved by updating the firmware?

Hello @kairenw . This error doesn't relate to the `initialization error` before. This error happens when my `eigen3` version is 3.3.9. After I reinstall `eigen3` of version 3.3.7. It was solved.

But actually, I'm not sure if this is truly the reason.