ouster-lidar/ouster-sdk #294: launch node crashes after initialising successfully,Exiting because of failure to write metadata path.

octobotics, 2021-08-24:
setting /run_id to b4e5e8a4-04d6-11ec-8037-00044be59260
process[rosout-1]: started with pid [14122]
started core service [/rosout]
process[os_node-2]: started with pid [14131]
process[os_cloud_node-3]: started with pid [14138]
process[relay_lidar-4]: started with pid [14147]
process[relay_imu-5]: started with pid [14155]
[ INFO] [1629808077.200574366]: Client version: 0.2.1+47f25ed-dirty-release
[ INFO] [1629808077.209317635]: Using automatic UDP destination
[ INFO] [1629808077.209806156]: Waiting for sensor os-992123000252.local to initialize ...
[ INFO] [1629808094.701682232]: Sensor initialized successfully
[ WARN] [1629808094.703786333]: Failed to write metadata to ouster_example/ouster_ros/meta_data/ouster_data.json; check that the path is valid. If you provided a relative path, please note that the working directory of all ROS nodes is set by default to $ROS_HOME
[ERROR] [1629808094.704368450]: Exiting because of failure to write metadata path
[ERROR] [1629808094.710882045]: Calling config service failed
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 14131, exit code 1, cmd /home/octo/catkin_ws/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/octo/.ros/log/b4e5e8a4-04d6-11ec-8037-00044be59260/os_node-2.log].
log file: /home/octo/.ros/log/b4e5e8a4-04d6-11ec-8037-00044be59260/os_node-2*.log
Initiating shutdown!
================================================================================
[relay_imu-5] killing on exit
[relay_lidar-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kairenw, 2021-08-24:
Hi @octobotics,

Could you show the whole command you're using to launch, and also double-check that your provided path is valid ?

kairenw, 2021-09-10:
It looks from the error like octobotics was providing the metadata path of `ouster_example/ouster_ros/meta_data/ouster_data.json`, actually. Not sure if the path is valid, however.

Closing because of lack of response. Feel free to reopen, @octobotics, if you still have questions.

closed 2021-09-10 by kairenw