ouster-lidar/ouster-sdk #296: ros topic delay increasing with time

Jaetriel, 2021-09-03:
Hi, I've got an OS1-128 using the latest version of the firmware. Running the ROS node at 2048x10 mode synced with a Jetson AGX via hardware PTP, the results of `rostopic delay /os_cloud_node/points` command start from around 0.140 sec, but over the course of about 10-15 mins slowly progress to 0.250~ (which feels like quite a lot) and then it seems like they kinda slowly fluctuate between the two. What are the possible reasons and is there a way to make the delay more stable?

dgarbuzov, 2021-12-29:
That could be the ROS message queue backing up due to processing falling behind. First thing to check might be that you're building in Release (not Debug) mode. If that's not it, could you share the CPU utilization of the ROS nodes?

Samahu, 2022-08-05:
@Jaetriel based on our internal analysis and current published code. A 250 ms is the expected delay for `/os_cloud_node/points` topic when using release build and the **2048x10** lidar mode.

With current code the point cloud message delay can range between 200-350 ms. ROS 1 is not a RTOS and thus message delay fluctuations are inherent in its design.

We do have an upcoming update that makes use of **ROS Nodelets**, which allows messages to bypass the network communication and copy the messages directly when possible (i.e. when running the nodelets within the same host). This will reduce the overall delay but won't get rid of the fluctuation due to the nature of ROS. It will, however, reduce the fluctuation and improve stability.

Please keep an eye on the next update to our ROS driver.

closed 2022-08-08 by Samahu

mertyavuz41, 2023-10-05:
Hi @Samahu,

I'm typing here because we are encountered same issue. I would like to ask that if we changed/decrease LiDAR's resolution mode, is the LiDAR's delay will be normal or significantly decreased? 

Thanks,