ouster-lidar/ouster-sdk #304: manifold2C++++process has died [pid 2224, exit code -6, cmd /home/dji/ouster32_ws/devel/lib/ouster_ros/os_cloud_node

zhangyuan12121, 2021-10-04:
I ran this driver on DJI's onboard processor (manifold2), and the compilation was normal, but an error was reported after running.

What is the problem?
[image]

[image]

ubuntu16.04
ros: kinetic

`Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:44993/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 
 * /os_node/lidar_port: 0
 * /os_node/metadata: test.json
 * /os_node/replay: False
 * /os_node/sensor_hostname: 192.168.1.200
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 192.168.1.102
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [2207]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 938df7a8-24c7-11ec-b841-34d262856487
process[rosout-1]: started with pid [2220]
started core service [/rosout]
process[os_node-2]: started with pid [2223]
process[os_cloud_node-3]: started with pid [2224]
process[relay_lidar-4]: started with pid [2225]
process[relay_imu-5]: started with pid [2226]
process[rviz-6]: started with pid [2227]
[ INFO] [1633320015.526410599]: Client version: 0.2.1+47f25ed-dirty-release
[ INFO] [1633320015.526503985]: Sending UDP data to 192.168.1.102
[ INFO] [1633320015.526530534]: Waiting for sensor 192.168.1.200 to initialize ...
[ INFO] [1633320025.376086189]: Sensor initialized successfully
[ INFO] [1633320025.377133979]: Wrote metadata to test.json
[ INFO] [1633320025.378396023]: Using lidar_mode: 1024x10
[ INFO] [1633320025.378510400]: OS-1-32-U2 sn: 122119000305 firmware rev: v2.0.0
*** Error in `/home/dji/ouster32_ws/devel/lib/ouster_ros/os_cloud_node': double free or corruption (out): 0x0000000001073aa0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777f5)[0x7fd55effd7f5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8038a)[0x7fd55f00638a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fd55f00a58c]
/home/dji/ouster32_ws/devel/lib/libouster_ros.so(_ZN10ouster_ros13scan_to_cloudERKN6ouster6XYZLutENSt6chrono8durationIlSt5ratioILl1ELl1000000000EEEERKNS0_9LidarScanERN3pcl10PointCloudINS_5PointEEE+0x270)[0x7fd5605fbf60]
/home/dji/ouster32_ws/devel/lib/ouster_ros/os_cloud_node[0x409011]
/home/dji/ouster32_ws/devel/lib/ouster_ros/os_cloud_node(_ZN3ros27SubscriptionCallbackHelperTIRKN10ouster_ros10PacketMsg_ISaIvEEEvE4callERNS_36SubscriptionCallbackHelperCallParamsE+0x2e6)[0x40ed56]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros17SubscriptionQueue4callEv+0xadd)[0x7fd5600ebc8d]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue9callOneCBEPNS0_3TLSE+0x4b0)[0x7fd5600919e0]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue13callAvailableENS_12WallDurationE+0x54b)[0x7fd56009341b]
[... 180 lines cut ...]
7fd5609ec000-7fd560a0b000 rw-p 00000000 00:00 0 
7fd560a3a000-7fd560a4d000 rw-p 00000000 00:00 0 
7fd560a4d000-7fd560a4e000 r--p 00025000 08:02 1324748                    /lib/x86_64-linux-gnu/ld-2.23.so
7fd560a4e000-7fd560a4f000 rw-p 00026000 08:02 1324748                    /lib/x86_64-linux-gnu/ld-2.23.so
7fd560a4f000-7fd560a50000 rw-p 00000000 00:00 0 
7fff7b374000-7fff7b396000 rw-p 00000000 00:00 0                          [stack]
7fff7b3de000-7fff7b3e1000 r--p 00000000 00:00 0                          [vvar]
7fff7b3e1000-7fff7b3e3000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
================================================================================REQUIRED process [os_cloud_node-3] has died!
process has died [pid 2224, exit code -6, cmd /home/dji/ouster32_ws/devel/lib/ouster_ros/os_cloud_node ~/os_config:=/os_node/os_config ~/lidar_packets:=/os_node/lidar_packets ~/imu_packets:=/os_node/imu_packets __name:=os_cloud_node __log:=/home/dji/.ros/log/938df7a8-24c7-11ec-b841-34d262856487/os_cloud_node-3.log].
log file: /home/dji/.ros/log/938df7a8-24c7-11ec-b841-34d262856487/os_cloud_node-3*.log
Initiating shutdown!
================================================================================
[rviz-6] killing on exit
[relay_imu-5] killing on exit
[relay_lidar-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[ INFO] [1633320026.071683232]: poll_client: caught signal, exiting
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

snakehaihai, 2021-10-16:
Another thing i notices is that you seems to have corruption when running. Try to delete the devel and build. then catkin_make again to see if you can get rid of that currption error

bexcite, 2021-12-14:
Hi @zhangyuan12121,

Double free corruption maybe related to a missing compile definition EIGEN_MAX_ALIGN_BYTES=32
(related https://github.com/ouster-lidar/ouster_example/issues/299#issuecomment-924050547) that `ouster_ros` has by default.