ouster-lidar/ouster-sdk #311: Namespacing Ouster ROS launch file

derektan95, 2021-10-20:
Hi, 

I am currently trying to namespace the entire ouster.launch in the ouster_example folder. When I ran `export ROS_NAMESPACE=<ROBOT_NAMESPACE>`, followed by `roslaunch ouster_ros ouster.launch`, the pointcloud topic `/os_cloud_node/points` mysteriously disappears.

I wonder if you have any insights to this? I would like to have all of my ouster nodes and topics namespaced via a convenient way. Thanks. 

Regards,
Derek

its-pru, 2022-07-01:
I am running into the same problem with name spacing . I am trying to run two ousters at the same time but when I make groups in the launch file for the different devices the pointcloud topic /os_cloud_node/point disappears. Any help would be greatly appreciated.

Samahu, 2022-07-01:
We made changes to enclose all of our nodes with a specific namespace, but we haven't push the changes yet. Please keep an eye on this on our next update.

Samahu, 2022-08-24:
@derektan95 and @its-pru A new release has been published which arranges all created nodes and topics under a single configurable ROS namespace. Please refer to https://github.com/ouster-lidar/ouster_example/pull/420 for details.

closed 2022-08-24 by Samahu