ouster-lidar/ouster-sdk #316: convert pcap to rosbag

tianshiz, 2021-10-30:
I see in the sdk that we can convert a pcap to csv. But is there a way to do this and convert to a rosbag with sensor_msg/PointCloud2 and sensor_msg/imu msg format?
If not, how do we take the row by row extracted in the csv demo and extract the imu and lidar data separately?

kairenw, 2021-11-29:
Hi @tianshiz,

Unfortunately, we do not currently support converting pcaps to rosbag with those message types. I recommend using the `ouster_ros` driver if you wish to capture bags.

The csv demo only records the lidar data currently. If you'd like to batch lidar packets and capture imu packets, and record their values to csv, I suggest that you work with the packets directly using a `PacketSource` on the python side.

Ali-Darzi, 2022-06-01:
Conversion of  Pcap to rosbag is doable using [this code](https://github.com/Krishtof-Korda/pcap-to-bag) 
to use it, clone the repo to the source folder of you Catkin workspace, the same workspace as ouster_example. After building, use ./pcap_to_bag to convert pcap to rosbag. 
To keep the output file size as small as possible, the output of os_node (which are imu_packets and lidar_packets) can be found in the rosbag file. To have the output of os_cloud_node (imu and PointCloud), the ouster_ros master should be running on another terminal. It will act as a translator that listens to the os_node.