ouster-lidar/ouster-sdk #321: No data received in docker container

themightyoarfish, 2021-11-23:
I'm running the ROS nodes inside docker (with `--net=host`), with the scanner connected directly to the host machine. The initialization seems to be successful, but no data is published on the points topic.

```
root@docker-desktop:/autocrane-lidar/catkin_ws# roslaunch ouster_ros ouster.launch metadata:=wat.json sensor_hostname:=os-122142000686.local udp_dest:=192.168.65.3
... logging to /root/.ros/log/0e3027d4-4c4a-11ec-8bc2-025000000001/roslaunch-docker-desktop-808.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://docker-desktop:55321/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix:
 * /os_node/imu_port: 0
 * /os_node/lidar_mode:
 * /os_node/lidar_port: 0
 * /os_node/metadata: wat.json
 * /os_node/replay: False
 * /os_node/sensor_hostname: os-122142000686.l...
 * /os_node/timestamp_mode:
 * /os_node/udp_dest: 192.168.65.3
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)

ROS_MASTER_URI=http://localhost:11311

process[os_node-1]: started with pid [823]
[ INFO] [1637664702.986440927]: Client version: 0.3.0.dev2+3e6af59-release
[ INFO] [1637664702.988041368]: Sending UDP data to 192.168.65.3
[ INFO] [1637664702.988122757]: Waiting for sensor os-122142000686.local to initialize ...
process[os_cloud_node-2]: started with pid [824]
process[relay_lidar-3]: started with pid [830]
process[relay_imu-4]: started with pid [836]
[ INFO] [1637664720.647720148]: Sensor initialized successfully
[ INFO] [1637664720.649444786]: Wrote metadata to wat.json
WARNING: No lidar profile found.
[ INFO] [1637664720.650959480]: Using lidar_mode: 2048x10
[ INFO] [1637664720.651086717]: OS-1-128 sn: 122142000686 firmware rev: v2.1.3
```

themightyoarfish, 2021-11-23:
It turns out that `--net=host` does not work, you must explicitly forward ports

```
> docker run --rm --name os1-2 -p 7502:7502/udp -it ouster-ros sensor_hostname:=os-122142000686.local metadata:=wat.json  lidar_port:=7502
```

closed 2021-11-23 by themightyoarfish

ramandika, 2024-02-05:
Hey @themightyoarfish how do you get to access the ouster library in the docker container? I build and run the docker container but I can't access the library inside the container

```build@a1f14539f105:~$ python3
Python 3.8.10 (default, Nov 22 2023, 10:22:35) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ouster
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'ouster'
```

ramandika, 2024-02-05:
Doesn't run the docker build and start the container enough in this case?