ouster-lidar/ouster-sdk #323: lidar scan not complete

yinwu33, 2021-11-29:
[image]
[image]

Hello I recorded some rosbag in Ubuntu 20.04 based on this repo. But some frame seems not complete, only around 270 degree or less (in 2. image). Is this normal or I can do something to improve the scan?

kairenw, 2021-11-29:
Hi @yinwu33,

Could be a few things. If it's the last frame, then the frame may just be incomplete because the recording was stopped before the full frame could be captured. Otherwise, it may seem that a frame is incomplete due to minimum range. You might also wish to check your azimuth_window settings (easily accessed by going to your sensor's homepage in the browser). 

Hope that helps!

yinwu33, 2021-11-29:
> 

Thanks for your reply. It's not the last frame. I replayed the rosbag and viewed the scan with rviz. The scan seems to twinkle due to several frames are not complete. The incomplete frame actually twinkles frequently. What could I do with `azimuth_window` settings?

In addition, I ran 6 lidars in total to record data. Could this be a problem?

kairenw, 2021-11-29:
Hi @yinwu33, 

You should check in the sensor homepage that your azimuth_window settings are for the full window ( [0, 360000] ). 

It's possible you are experiencing packet loss. I think the best thing to do is to contact [our sensor support.](https://ouster.atlassian.net/servicedesk/customer/portal/8) so they can help you diagnose the issue.

yinwu33, 2021-12-03:
> Hi @yinwu33,
>
> You should check in the sensor homepage that your azimuth_window settings are for the full window ( [0, 360000] ).
>
> It's possible you are experiencing packet loss. I think the best [...]

Hello @kairenw 
Thanks for your advice. I checked the `azimuth_window` setting, which is set up correctly. However, I have another guess. Since we have installed 6 Ouster lidars in our vehicle, and use a `ros launch file` to start six nodes of yours. Would this be a reason due to the huge amount of data?

kairenw, 2021-12-03:
Hi @yinwu33,

Great, I'm glad it's not the azimuth window. 

It's definitely possible that you're getting packet loss due to having a couple sensors. Our client and viz code is performant enough to support 6 sensors, but the devil is really in the details. I'd contact [Ouster support ](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) with the code you're using, and a recorded bag of the data (along with the metadata) so we can look at it on our end.

Thanks!
Kairen

kairenw, 2021-12-03:
Going to close this out now. In your support ticket, you can also link to this GH issue.

closed 2021-12-03 by kairenw

themightyoarfish, 2023-11-08:
Is there sample code how to identify when a scan is incomplete? Especially in the presence of a limited azimuth window?

Samahu, 2023-11-08:
You can use the `LidarScan::complete()` method to check if a LidarScan is complete over a specified azimuth window:
```cpp
bool LidarScan::complete(sensor::ColumnWindow window) const
```