ouster-lidar/ouster-sdk #324: jetson xavier nx + OS0 lidar

junjie-Hub, 2021-12-02:
Hi! I want to configure ouster os0 on jetson xavier nx, is there any solution?

junjie-Hub, 2021-12-02:
when I used commend:nmcli con modify eth1 ipv4.method link-local ipv4.addresses "
[image]
But if it is correct, the following result should appear.
[image]

 So i need your help.......

kairenw, 2021-12-03:
Hi @junjie-Hub, Think it's likely you don't have the interface `eth1`. You can identify the interface by using `ifconfig`. Try following the directions under 17.3.4 "Determining the IPv4 Address of the Interface" in our [user manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2.1.x.pdf)

junjie-Hub, 2021-12-06:
> 

OK. Thank you for your response. I will try as you said.

kairenw, 2021-12-14:
Let us know if you have other issues. Closing ticket now.

closed 2021-12-14 by kairenw