ouster-lidar/ouster-sdk #326: Python example code does not work

themightyoarfish, 2021-12-11:
I have an OS1 sensor, and `pip install`ed the ouster sdk. Some examples work:

```
> python -m ouster.sdk.examples.client  os-122142000686.local fetch-metadata
example: fetch-metadata
Retrieved metadata:
  serial no:        122142000686
  firmware version: v2.1.3
  product line:     OS-1-128
  lidar mode:       2048x10
Writing to: os-122142000686.local.json
```

But actually retrieving data does not. I can successfully use the cpp client on the same computer (macos 10.15.7).

```
> python -m ouster.sdk.examples.client  os-122142000686.local plot-xyz-points
example: plot-xyz-points
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.8/3.8.8_1/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/Cellar/python@3.8/3.8.8_1/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/rasmus/Documents/Psiori/Projects/AutoCrane/autocrane-core/autocrane_venv/lib/python3.8/site-packages/ouster/sdk/examples/client.py", line 416, in <module>
    main()
  File "/Users/rasmus/Documents/Psiori/Projects/AutoCrane/autocrane-core/autocrane_venv/lib/python3.8/site-packages/ouster/sdk/examples/client.py", line 412, in main
    example(args.hostname)  # type: ignore
  File "/Users/rasmus/Documents/Psiori/Projects/AutoCrane/autocrane-core/autocrane_venv/lib/python3.8/site-packages/ouster/sdk/examples/client.py", line 245, in plot_xyz_points
    scan = next(sample)[0]
  File "/Users/rasmus/Documents/Psiori/Projects/AutoCrane/autocrane-core/autocrane_venv/lib/python3.8/site-packages/ouster/client/core.py", line 405, in next_batch
    source.flush(full=True)
  File "/Users/rasmus/Documents/Psiori/Projects/AutoCrane/autocrane-core/autocrane_venv/lib/python3.8/site-packages/ouster/client/core.py", line 254, in flush
    raise ClientTimeout(
ouster.client.core.ClientTimeout: No packets received within 1.0s
```

themightyoarfish, 2021-12-11:
Changing the timeout does not help, I can tell the scanner is not starting as it does with the C++ code.

themightyoarfish, 2021-12-11:
The scanner is connected via Link Local address, and should use the default ports.

kairenw, 2021-12-11:
Hi @themightyoarfish, 

Any chance you haven't set the udp_dest? Try going to the sensor homepage: http:<sensor-hostname.local>/config and using "set local" (remember to "Apply Config" after that).

themightyoarfish, 2021-12-11:
The C++ code deals without, I had assumed the Python code to behave the same and the sensor to send to whatever origin address it received commands from.

kairenw, 2021-12-13:
Yes, taht's correct. The Python `plot-xyz-points` example does not auto-configure udp_dest, but the cpp example `ouster_client_example` does. Let me know if that solves your issue or if it remains.

dgarbuzov, 2021-12-28:
Think the only reason the examples (other than `configure-sensor`) don't change the configuration is that it's annoying to wait for the sensor to reinitialize on every run. The intention was that people would go through the examples section of the docs top-to-bottom, but we should at least make that more clear.

themightyoarfish, 2021-12-29:
Wait, so the long startup time is because of this? That has been an annoyance with my c++ client code also.

dgarbuzov, 2022-01-05:
Yep. There's two overloads of `init_client()` -- one updates a few config parameters and reinitializes the sensor, the other skips that step and immediately fetches metadata and begins listening for UDP packets.  The example code uses the former.

sradmard, 2022-01-05:
I was running into the same issue, and setting `udp_dest` resolved the issue for me as recommended.

kairenw, 2022-02-03:
Think this is probably resolved. Closing now.

kairenw, 2022-02-03:
Reopning to note that the requested enhancement is: rework examples section of docs so that it's clear you need to configure the sensors first before running some of the examples.

kairenw, 2022-02-08:
Issue should be addressed. Thank you!

closed 2022-02-08 by kairenw

MarcoCia, 2022-07-12:
> Stavo riscontrando lo stesso problema e l'impostazione `udp_dest`mi ha risolto il problema come consigliato.

Hi, Can you help me to solve the problem? 
I receive the error "ouster.client.core.ClientTimeout: No packets received within 1.0s"

kairenw, 2022-07-12:
Sure, could you go to your sensor homepage (http://[SENSOR_HOSTNAME].local) and then hit the "Configuration tab" and upload a screenshot?

will22thomas, 2023-03-14:
I'm having a similar problem: every now and then it really acts up and I have to unplug the lidar from the wall, or the computer, and re-configure it, and reset the UDP destination address (or any combination of these solutions). Sometimes it eventually works, sometimes it doesn't. I get this error: ouster.client.core.ClientTimeout: No packets received within 2.0s
Any ideas?

kairenw, 2023-03-14:
hi @MarcoCia and @will22thomas in general it's a bit easier for us if you start a new issue. Just tag me on the new one and we can talk there (:

sriram2989, 2023-03-21:
same issue with by any chance can i have a solution