ouster-lidar/ouster-sdk #327: Failed to find match for field 'ring'.

avinash-yadav1, 2021-12-12:
Hi we are using OS-1 64 channel lidar and when we are trying to process the lidar point cloud data through our calibration algorithm then we are getting this error:**Failed to find match for field 'ring'.* * But i'm not seeing this error with the velodyne lidar.  Please help us on this issue.

themightyoarfish, 2021-12-12:
The ouster point cloud does not have a "ring" field, while your velodyne driver probably outputs one. You'd need to add your own code to the conversion from `ouster::LidarScan` to ROS pointcloud to add this field.

kairenw, 2021-12-13:
Essentially, however you're connecting your calibration algorithm to ours, you need to provide the 'ring' field first. 'Ring' is the same as the row index in an `ouster::LidarScan`. It may be helpful to look at [how we populate the ring](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/ros.cpp#L87) in `ouster_ros`. 

Closing this ticket now. If you're having trouble connecting them and don't want to share your code in public, it may be easier to contact us via [Ouster customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8).

closed 2021-12-13 by kairenw