ouster-lidar/ouster-sdk #328: Lidar failed receiving time from PTP, ETHERNET_LINK_BAD alert

AndreasInSchwaben, 2021-12-13:
Hello,

I tried to setup my Lidars to by synced via PTP.

`get_time_info` shows
```
{"timestamp": {"time": 3818.798550643, "mode": "TIME_FROM_PTP_1588", "time_options": {"ptp_1588": 3818, "sync_pulse_in": 0, "internal_osc": 3810}}, "sync_pulse_in": {"locked": 0, "polarity": "ACTIVE_HIGH", "diagnostics": {"last_period_nsec": 0, "count": 0, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "OFF", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 0, "polarity": "ACTIVE_HIGH", "ignore_valid_char": 0, "baud_rate": "BAUD_9600", "leap_seconds": 0, "diagnostics": {"decoding": {"utc_decoded_count": 0, "date_decoded_count": 0, "not_valid_count": 0, "last_read_message": ""}, "io_checks": {"start_char_count": 0, "char_count": 0, "bit_count": 1, "bit_count_unfiltered": 0}}}}}
```
So the mode was set to successfully to `TIME_FROM_PTP_1588`.
The time value `3818.798550643` is the time in sec since power-on the Lidar. If PTP would run correctly I would expect to see a time like `1639390007.73391914` since 1. Jan 1970.

On my PC a PTP daemon was started. His log-file shows:
```
2021-12-13 07:29:57.841956 ptpd2[2227].startup (info)      (___)   Info:    Now running as a daemon
2021-12-13 07:29:57.844580 ptpd2[2227].startup (notice)    (___) PTPDv2 started successfully on igb2 using "masteronly" preset (PID 2227)
2021-12-13 07:29:57.844605 ptpd2[2227].startup (info)      (___) TimingService.PTP0: PTP service init
2021-12-13 07:29:57.847242 ptpd2[2227].igb2 (warning)   (init) Interface igb2 seems to be down. PTPd will not operate correctly until it's up.
2021-12-13 07:29:57.948409 ptpd2[2227].igb2 (notice)    (lstn_init) Now in state: PTP_LISTENING
2021-12-13 07:30:07.844666 ptpd2[2227].igb2 (notice)    (lstn_init) TimingService.PTP0: elected best TimingService
2021-12-13 07:30:07.844732 ptpd2[2227].igb2 (info)      (lstn_init) TimingService.PTP0: acquired clock control
2021-12-13 07:30:09.949225 ptpd2[2227].igb2 (notice)    (mst) Now in state: PTP_MASTER, Best master: 00e04bfffe611357(unknown)/1 (self)
```
I expect this to be setup successful. At least, this works for other sensors.

However, wireshark monitors only the `Announce Message`, `Sync Message` and `Follow_Up Message`. There is no PTP reply from Lidars visible.

The PC and Lidars are connected through a Netgear GS110EMX-100PES. This is a switch with GbE and 10GbE ports.

The Lidar shows an alert when
`get_alerts`
```
{"log": [{"active": true, "category": "ETHERNET_LINK_BAD", "cursor": 0, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 0/Unknown", "realtime": "27286386147"}, {"active": true, "category": "MOTOR_CONTROL", "cursor": 1, "id": "0x01000050", "level": "WARNING", "msg": "The phase lock offset error has exceeded the threshold.", "msg_verbose": "", "realtime": "32429986594"}, {"active": true, "category": "INTERNAL_FAULT", "cursor": 2, "id": "0x0100005a", "level": "WARNING", "msg": "Unit has experienced an internal COMM warning: some measurements may have been skipped.", "msg_verbose": "", "realtime": "33430391358"}, {"active": false, "category": "MOTOR_CONTROL", "cursor": 3, "id": "0x01000050", "level": "WARNING", "msg": "The phase lock offset error has exceeded the threshold.", "msg_verbose": "", "realtime": "33430778633"}, {"active": true, "category": "UDP_TRANSMISSION", "cursor": 4, "id": "0x01000016", "level": "WARNING", "msg": "Could not send lidar data UDP packet to host; check that network is up.", "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.178.23:7504", "realtime": "35600679746"}, {"active": false, "category": "ETHERNET_LINK_BAD", "cursor": 5, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 1000/Full", "realtime": "36659599899"}, {"active": true, "category": "INTERNAL_FAULT", "cursor": 6, "id": "0x0100005b", "level": "WARNING", "msg": "Unit has experienced an internal COMM warning: some measurements may have been skipped.", "msg_verbose": "", "realtime": "37430848588"}, {"active": true, "category": "UDP_TRANSMISSION", "cursor": 7, "id": "0x01000019", "level": "WARNING", "msg": "Could not send IMU UDP packet to host; check that network is up.", "msg_verbose": "Failed to send imu UDP data to destination host 192.168.178.23:7505", "realtime": "38726319922"}, {"active": false, "category": "UDP_TRANSMISSION", "cursor": 8, "id": "0x01000016", "level": "WARNING", "msg": "Could not send lidar data UDP packet to host; check that network is up.", "msg_verbose": "Cleared by reinitialization.", "realtime": "49386245219"}, {"active": false, "category": "UDP_TRANSMISSION", "cursor": 9, "id": "0x01000019", "level": "WARNING", "msg": "Could not send IMU UDP packet to host; check that network is up.", "msg_verbose": "Cleared by reinitialization.", "realtime": "49386540286"}, {"active": false, "category": "INTERNAL_FAULT", "cursor": 10, "id": "0x0100005a", "level": "WARNING", "msg": "Unit has experienced an internal COMM warning: some measurements may have been skipped.", "msg_verbose": "Cleared by reinitialization.", "realtime": "49386761586"}, {"active": false, "category": "INTERNAL_FAULT", "cursor": 11, "id": "0x0100005b", "level": "WARNING", "msg": "Unit has experienced an internal COMM warning: some measurements may have been skipped.", "msg_verbose": "Cleared by reinitialization.", "realtime": "49386964444"}, {"active": true, "category": "UDP_TRANSMISSION", "cursor": 12, "id": "0x01000018", "level": "WARNING", "msg": "Client machine announced it is not reachable on the provided not reachable on IMU data port; check that udp_dest and udp_port_imu configured on the sensor matches client IP and port.", "msg_verbose": "Failed to send imu UDP data to destination host 192.168.214.2:42427", "realtime": "11112142233187"}, {"active": true, "category": "UDP_TRANSMISSION", "cursor": 13, "id": "0x01000015", "level": "WARNING", "msg": "Client machine announced it is not reachable on the provided lidar data port; check that udp_dest and udp_port_lidar configured on the sensor matches client IP and port.", "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.214.2:37808", "realtime": "11113133454004"}], "next_cursor": 14, "active": [{"active": true, "category": "UDP_TRANSMISSION", "cursor": 13, "id": "0x01000015", "level": "WARNING", "msg": "Client machine announced it is not reachable on the provided lidar data port; check that udp_dest and udp_port_lidar configured on the sensor matches client IP and port.", "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.214.2:37808", "realtime": "11113133454004"}, {"active": true, "category": "UDP_TRANSMISSION", "cursor": 12, "id": "0x01000018", "level": "WARNING", "msg": "Client machine announced it is not reachable on the provided not reachable on IMU data port; check that udp_dest and udp_port_imu configured on the sensor matches client IP and port.", "msg_verbose": "Failed to send imu UDP data to destination host 192.168.214.2:42427", "realtime": "11112142233187"}]}
```
In spite of this `ETHERNET_LINK_BAD` alert, the point cloud data from multiple Lidar arrive correctly with more than 500MBit/s.
So I do not know if this alert related to the non-successful PTP sync.

There are also alerts complaining `Could not send lidar data UDP packet to host; check that network is up.` PTP is based on UDP messages. The Lidar fails to `send lidar UDP data to destination host 192.168.178.23:7504` 
The IP Address `192.168.178.23` does not exist in my network.

I also played setting the UDP address to a valid one:
```
set_config_param udp_dest 192.168.214.2
reinitialize
get_config_txt
{"udp_ip": "192.168.214.2", "udp_dest": "192.168.214.2", "udp_port_lidar": 37808, "udp_port_imu": 42427, "timestamp_mode": "TIME_FROM_PTP_1588", "sync_pulse_in_polarity": "ACTIVE_HIGH", "nmea_in_polarity": "ACTIVE_HIGH", "nmea_ignore_valid_char": 0, "nmea_baud_rate": "BAUD_9600", "nmea_leap_seconds": 0, "multipurpose_io_mode": "OFF", "sync_pulse_out_polarity": "ACTIVE_HIGH", "sync_pulse_out_frequency": 1, "sync_pulse_out_angle": 360, "sync_pulse_out_pulse_width": 10, "auto_start_flag": 1, "operating_mode": "NORMAL", "lidar_mode": "2048x10", "azimuth_window": [0, 360000], "phase_lock_enable": true, "phase_lock_offset": 180000}
```
But the alerts stay. PTP does not work.

Can anybody  help, please, why PTP is not syncing the clock on the Lidar?

Thanks!
Andreas

kairenw, 2021-12-13:
Hi @AndreasInSchwaben, 

Would you mind submitting the same issue through our customer service portal [here](https://ouster.atlassian.net/servicedesk/customer/portal/8)? As this pertains to our PTP functionality and not specifically the SDK code, you'll get a faster answer through there. I'll leave this open for now, and you can let me know if you don't get any response there.

AndreasInSchwaben, 2021-12-14:
Hi,

there was a wrong subdomain number set in our PTP messages. 
The subdomain has to be “0” in order Lidar accepting the PTP messages.

Best Regadrs
Andreas

kairenw, 2021-12-14:
Awesome, thanks for coming back with an update. Will close ticket now.

closed 2021-12-14 by kairenw