ouster-lidar/ouster-sdk #330: only partial data

liubonan123, 2021-12-15:


liubonan123, 2021-12-15:
[image]

liubonan123, 2021-12-15:
[image]

liubonan123, 2021-12-15:
[image]

kairenw, 2021-12-15:
Hi @liubonan123,

You need to set the azimuth window to end at 360000, not 360. Thanks! 

Closing ticket now.

liubonan123, 2021-12-16:
@kairenw 
after I set the azimuth window to end at 360000, I still have the same problem, and I can only get the msg every once in a while

{
  "alerts": {
    "active": [
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 10,
        "id": "0x01000015",
        "level": "WARNING",
        "msg": "Client machine announced it is not reachable on the provided lidar data port; check that udp_dest and udp_port_lidar configured on the sensor matches client IP and port.",
        "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.100.1:37443",
        "realtime": "889142716561"
      },
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 8,
        "id": "0x01000016",
        "level": "WARNING",
        "msg": "Could not send lidar data UDP packet to host; check that network is up.",
        "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.100.1:37443",
        "realtime": "827648123433"
      },
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 9,
        "id": "0x01000018",
        "level": "WARNING",
        "msg": "Client machine announced it is not reachable on the provided not reachable on IMU data port; check that udp_dest and udp_port_imu configured on the sensor matches client IP and port.",
        "msg_verbose": "Failed to send imu UDP data to destination host 192.168.100.1:59541",
        "realtime": "841149872468"
      },
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 7,
        "id": "0x01000055",
        "level": "WARNING",
        "msg": "Unit has experienced a packet drop rate above normal threshold. Please check that the network has at least 1000 Mbps connection. Common causes of this notice may be 100 or 10 Mbps network connections. ",
        "msg_verbose": "",
        "realtime": "799128911875"
      },
      {
        "active": true,
        "category": "INTERNAL_FAULT",
        "cursor": 6,
        "id": "0x0100005b",
        "level": "WARNING",
        "msg": "Unit has experienced an internal COMM warning: some measurements may have been skipped.",
        "msg_verbose": "",
        "realtime": "798056050326"
      }
    ],
    "log": [
      {
        "active": true,
        "category": "ETHERNET_LINK_BAD",
        "cursor": 0,
        "id": "0x01000011",
        "level": "WARNING",
        "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.",
        "msg_verbose": "Link transitioned to 0/Unknown",
        "realtime": "202965587807"
      },
      {
        "active": false,
[... 116 lines cut ...]
    "signal_multiplier": 1,
    "sync_pulse_in_polarity": "ACTIVE_HIGH",
    "sync_pulse_out_angle": 360,
    "sync_pulse_out_frequency": 1,
    "sync_pulse_out_polarity": "ACTIVE_HIGH",
    "sync_pulse_out_pulse_width": 10,
    "timestamp_mode": "TIME_FROM_INTERNAL_OSC",
    "udp_dest": "192.168.100.1",
    "udp_ip": "192.168.100.1",
    "udp_port_imu": 59541,
    "udp_port_lidar": 37443
  },
  "info": {
    "base_pn": "830-102905-E",
    "base_sn": "102141002404",
    "build_date": "2021-10-21T19:27:29Z",
    "build_rev": "v2.1.3",
    "image_rev": "ousteros-image-prod-aries-v2.1.3+20211021192601.patch-v2.1.3",
    "prod_line": "OS-1-128",
    "prod_pn": "840-102145-05",
    "prod_sn": "122146000203",
    "proto_rev": "v1.1.1",
    "status": "RUNNING"
  }
}

kairenw, 2021-12-16:
Hi @liubonan123,

`"msg": "Client machine announced it is not reachable on the provided lidar data port; check that udp_dest and udp_port_lidar configured on the sensor matches client IP and port.",
"msg_verbose": "Failed to send lidar UDP data to destination host 192.168.100.1:37443",` is probably the relevant section of the alerts.

As this is a networking issue, I'm going to direct you to the[ Ouster customer support portal](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86). 

Closing issue again.

closed 2021-12-16 by kairenw