ouster-lidar/ouster-sdk #332: Failed building wheel for ouster-sdk

Shibaditya99, 2021-12-16:
I used this command "py -3 -m pip install ouster-sdk[examples]" to install ouster-sdk in pycharm in the venv. But unfortunately it's showing  **Failed building wheel for ouster-sdk** this error. Cmake, pybind11 is already install and the pybind11 has the pybind11Config.cmake file I checked. Full error msg is below. It's showing the same error in power shell also.
`  ERROR: Command errored out with exit status 1:
   command: 'C:\Python310\python.exe' 'C:\Python310\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' build_wheel 'C:\Users\INX-DOC\AppData\Local\Temp\t
mpvruem1hd'
       cwd: C:\Users\INX-DOC\AppData\Local\Temp\pip-install-0qftsl7j\ouster-sdk_f0f5183682ce45caa5ed361b08941d09
  Complete output (107 lines):
  running bdist_wheel
  running build
  running build_py
  package init file 'src\cpp\__init__.py' not found (or not a regular file)
  package init file 'src\ouster\__init__.py' not found (or not a regular file)
  creating build
  creating build\lib.win-amd64-3.10
  creating build\lib.win-amd64-3.10\ouster
  creating build\lib.win-amd64-3.10\ouster\client
  copying src\ouster\client\core.py -> build\lib.win-amd64-3.10\ouster\client
  copying src\ouster\client\data.py -> build\lib.win-amd64-3.10\ouster\client
  copying src\ouster\client\_bufstream.py -> build\lib.win-amd64-3.10\ouster\client
  copying src\ouster\client\_digest.py -> build\lib.win-amd64-3.10\ouster\client
  copying src\ouster\client\__init__.py -> build\lib.win-amd64-3.10\ouster\client
  creating build\lib.win-amd64-3.10\ouster\pcap
  copying src\ouster\pcap\pcap.py -> build\lib.win-amd64-3.10\ouster\pcap
  copying src\ouster\pcap\__init__.py -> build\lib.win-amd64-3.10\ouster\pcap
  creating build\lib.win-amd64-3.10\ouster\sdk
  copying src\ouster\sdk\__init__.py -> build\lib.win-amd64-3.10\ouster\sdk
  creating build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\sdk\examples\client.py -> build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\sdk\examples\colormaps.py -> build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\sdk\examples\open3d.py -> build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\sdk\examples\pcap.py -> build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\sdk\examples\reference.py -> build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\sdk\examples\__init__.py -> build\lib.win-amd64-3.10\ouster\sdk\examples
  copying src\ouster\client\py.typed -> build\lib.win-amd64-3.10\ouster\client
  copying src\ouster\pcap\py.typed -> build\lib.win-amd64-3.10\ouster\pcap
  running build_ext
  -- Building for: Visual Studio 16 2019
  -- Ouster SDK location: C:/Users/INX-DOC/AppData/Local/Temp/pip-install-0qftsl7j/ouster-sdk_f0f5183682ce45caa5ed361b08941d09/sdk
  -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
  -- The C compiler identification is MSVC 19.29.30136.0
  -- The CXX compiler identification is MSVC 19.29.30136.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped        
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  CMake Error at CMakeLists.txt:11 (find_package):
    By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
    asked CMake to find a package configuration file provided by "pybind11",
    but CMake did not find one.

    Could not find a package configuration file provided by "pybind11"
    (requested version 2.0) with any of the following names:

      pybind11Config.cmake
      pybind11-config.cmake

    Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
    "pybind11_DIR" to a directory containing one of the above files.  If
    "pybind11" provides a separate development package or SDK, be sure it has
    been installed.

  -- Configuring incomplete, errors occurred!
  See also "C:/Users/INX-DOC/AppData/Local/Temp/pip-install-0qftsl7j/ouster-sdk_f0f5183682ce45caa5ed361b08941d09/build/temp.win-amd64-3.10/Release/CMakeFiles/CMakeOut
put.log".
  Traceback (most recent call last):
    File "C:\Python310\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 349, in <module>
      main()
    File "C:\Python310\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 331, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "C:\Python310\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 248, in build_wheel
      return _build_backend().build_wheel(wheel_directory, config_settings,
    File "C:\Users\INX-DOC\AppData\Local\Temp\pip-build-env-yrh4r9u3\overlay\Lib\site-packages\setuptools\build_meta.py", line 230, in build_wheel
      return self._build_with_temp_dir(['bdist_wheel'], '.whl',
    File "C:\Users\INX-DOC\AppData\Local\Temp\pip-build-env-yrh4r9u3\overlay\Lib\site-packages\setuptools\build_meta.py", line 215, in _build_with_temp_dir
      self.run_setup()
    File "C:\Users\INX-DOC\AppData\Local\Temp\pip-build-env-yrh4r9u3\overlay\Lib\site-packages\setuptools\build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 122, in <module>
      setup(
    File "C:\Users\INX-DOC\AppData\Local\Temp\pip-build-env-yrh4r9u3\overlay\Lib\site-packages\setuptools\__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "C:\Python310\lib\distutils\core.py", line 148, in setup
      dist.run_commands()
    File "C:\Python310\lib\distutils\dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "C:\Python310\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "setup.py", line 115, in run
      super().run()
    File "C:\Users\INX-DOC\AppData\Local\Temp\pip-build-env-yrh4r9u3\overlay\Lib\site-packages\wheel\bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "C:\Python310\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Python310\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Python310\lib\distutils\command\build.py", line 135, in run
      self.run_command(cmd_name)
    File "C:\Python310\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Python310\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "setup.py", line 38, in run
      self.build_extension(ext)
    File "setup.py", line 84, in build_extension
      subprocess.check_call(['cmake', ext.sourcedir] + cmake_args,
    File "C:\Python310\lib\subprocess.py", line 369, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', 'C:\\Users\\INX-DOC\\AppData\\Local\\Temp\\pip-install-0qftsl7j\\ouster-sdk_f0f5183682ce45caa5ed361b08941d09', '-D
CMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\INX-DOC\\AppData\\Local\\Temp\\pip-install-0qftsl7j\\ouster-sdk_f0f5183682ce45caa5ed361b08941d09\\build\\lib.win-amd64-3.10\
\ouster\\', '-DPYTHON_EXECUTABLE=C:\\Python310\\python.exe', '-A', 'x64', '-DOUSTER_SDK_PATH=C:\\Users\\INX-DOC\\AppData\\Local\\Temp\\pip-install-0qftsl7j\\ouster-sd
k_f0f5183682ce45caa5ed361b08941d09\\sdk']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for ouster-sdk
Failed to build ouster-sdk
ERROR: Could not build wheels for ouster-sdk which use PEP 517 and cannot be installed directly`

dgarbuzov, 2021-12-28:
What are you using to install the build dependencies? Since pybind11 is the first call to `find_package` I'm guessing it's not specific to that package.

If you're using vcpkg, did you run `vcpkg integrate install` to make sure VS can find libraries installed through vcpkg?

https://vcpkg.readthedocs.io/en/latest/users/integration/

lachlanbrewster, 2022-01-18:
I also have this problem. I am not building this repo, just installing it via pip.

dgarbuzov, 2022-01-18:
What OS and which versions of Python and pip are you using? If `pip install ouster-sdk` is trying to build from the source distribution, you're likely either using an old version of pip or are on a platform that we haven't published a binary package.

lachlanbrewster, 2022-01-19:
I am on ubuntu 18.04, running on a Jetson Xavier, using python 3.6.

Upgraded pip and it installed successfully! Thank you for the tip.

closed 2022-01-29 by Shibaditya99